probe-rs/probe-rs #3109: Failure to reset STM32WL33CCVx

hgaiser, 2025-02-18:
### Describe the bug

I am trying to program a [STM32WL33C2 Nucleo](https://www.st.com/en/evaluation-tools/nucleo-wl33cc2.html) board, but probe-rs fails to reset.

### To Reproduce

1. Generate chip description with target-gen: `cargo run --release -- arm --filter STM32WL3`
2. Attempt to reset the board: `probe-rs reset --chip STM32WL33CCVx --chip-description-path STM32WL3_Series.yaml`.

<details>
<summary><b>STM32WL3_Series.yaml</b></summary>

```
name: STM32WL3 Series
manufacturer:
  id: 0x20
  cc: 0x0
generated_from_pack: true
pack_file_release: 1.0.0
variants:
- name: STM32WL33C8Vx
  cores:
  - name: main
    type: armv6m
    core_access_options: !Arm
      ap: !v1 0
  memory_map:
  - !Nvm
    name: Main_Flash
    range:
      start: 0x10040000
      end: 0x10050000
    cores:
    - main
    access:
      write: false
      boot: true
  - !Ram
    name: SRAM1
    range:
      start: 0x20000000
      end: 0x20004000
    cores:
    - main
  flash_algorithms:
  - stm32wl3x_256k
- name: STM32WL33CBVx
  cores:
  - name: main
    type: armv6m
    core_access_options: !Arm
      ap: !v1 0
  memory_map:
  - !Nvm
    name: Main_Flash
    range:
      start: 0x10040000
      end: 0x10060000
    cores:
    - main
    access:
      write: false
      boot: true
  - !Ram
    name: SRAM1
    range:
      start: 0x20000000
      end: 0x20008000
    cores:
    - main
[... 241 lines cut ...]

Stack backtrace:
   0: anyhow::error::<impl anyhow::Error>::msg
   1: probe_rs::util::cli::attach_probe::{{closure}}
   2: probe_rs::Cli::run::{{closure}}
   3: tokio::runtime::park::CachedParkThread::block_on
   4: probe_rs::main
   5: std::sys::backtrace::__rust_begin_short_backtrace
   6: std::rt::lang_start::{{closure}}
   7: std::rt::lang_start_internal
   8: main
   9: <unknown>
  10: __libc_start_main
  11: _start

After which I have to power cycle the board, otherwise I keep getting `JtagDbgPowerError` even without `--connect-under-reset`.
```

### Operating System

Other Linux Distribution

### Additional context

I am very new to embedded programming, so I could very well be making a beginners mistake. I also created a similar discussion topic [here](https://github.com/probe-rs/probe-rs/discussions/3027).

Yatekii, 2025-02-22:
Probably ARMv6 needs this fix: https://github.com/probe-rs/probe-rs/pull/2852 as well.

hgaiser, 2025-02-25:
> Probably ARMv6 needs this fix: [#2852](https://github.com/probe-rs/probe-rs/pull/2852) as well.

Thanks for your reply! I just tried to apply the fix from #2852 to stm32_armv6.rs, but no luck unfortunately (same result).

Here is the commit in case you're interested https://github.com/probe-rs/probe-rs/commit/e77ba11b18f5efadaff473ddda2dfe8a6222251b

If you have other suggestions, I'm all ears :).

hgaiser, 2025-02-25:
I noticed two interesting things:

1. This device is expected to be an [armv7](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/vendor/st/mod.rs#L41-L43) chip, but it is an armv6 chip. Maybe there are other chips that start with `STM32WL` that are armv7? I haven't checked that yet.
2. The existing register addresses for the specialized chip families didn't make sense for the STM32WL family. I modified the register addresses according to the [reference manual](https://www.st.com/resource/en/reference_manual/rm0511-stm32wl30xx31xx33xx-armbased-wireless-mcus-with-subghz-radio-solution-stmicroelectronics.pdf) and to the best of my abilities. But mostly, I was comparing it with an existing family ([STM32F0](https://www.st.com/resource/en/reference_manual/rm0091-stm32f0x1stm32f0x2stm32f0x8-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)) and making some educated guesses as to what things should be.

I committed my changes here: https://github.com/probe-rs/probe-rs/commit/2da003f5669a57f1e49773afe187d3b0f0427467 . I made the changes to work for my device, I will worry about support for other devices when I submit a PR :).

With these changes it creates the correct armv6 debug sequence, however it unfortunately still gives me the same output.

hgaiser, 2025-10-04:
I've found a workaround, though I'm not certain why it works. If I call this:

```sh
```

Yatekii, 2025-10-04:
Yes, this might well be. The debug peripheral does not get reset until you power cycle.

Might be worth it to dig into what openocd is doing. Also, even easier potentially is to look at the CMSIS-Pack sequence for that device. I am not sure if it's available.