probe-rs/probe-rs #3112: Flash content verification failed with probe-rs >= 0.25.0, Raspberry Pi Pico, and Raspberry Pi Debug Probe

tti0, 2025-02-20:
### Describe the bug

Using `probe-rs run` and `probe-rs download` with `--chip RP2040 --verify` fails, with the error:

```
Error: The flashing procedure failed for 'blink.elf'.

Caused by:
    Flash content verification failed.
```

The target hardware is a Raspberry Pi Pico v1 (RP2040) and Raspberry Pi Debug Probe, running factory firmware. This behaviour is seen with probe-rs versions 0.25.0, 0.26.0, and 0.27.0. The last version without this problem is 0.24.0.

Running `probe-rs run` or `probe-rs download` with `--chip RP2040` but without `--verify` is successful, and the binary runs on the Pi Pico.

### To Reproduce

With probe-rs >= 0.25.0 installed (using the install script) on Linux:

1. Follow instructions in the Raspberry Pi C++ SDK Getting Started guide to build a blink binary [here](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf#%5B%7B%22num%22%3A36%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C165.76213%2Cnull%5D) -- this produces `blink.elf`
2. Connect a Raspberry Pi Pico and Raspberry Pi Debug Probe to your PC via USB, and connect the probe's debug port to the Pico's SWD pins
3. Run `probe-rs download blink.elf --chip RP2040 --verify`: this fails
4. Run `probe-rs download blink.elf --chip RP2040`: this succeeds

Repeat the same steps with probe-rs 0.24.0: (3) and (4) both succeed

### Stacktrace

```shell
$ RUST_LOG=info probe-rs download blink.elf --chip RP2040 --verify
 INFO probe_rs::config::target: Using sequence Arm(DefaultArmSequence(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
 INFO probe_rs::architecture::arm::sequences: Debug port Multidrop(1002927) is powered down, powering up
 INFO probe_rs::architecture::arm::communication_interface: Searching valid APs
 INFO probe_rs::architecture::arm::sequences: Debug port Multidrop(11002927) is powered down, powering up
 INFO probe_rs::architecture::arm::communication_interface: Searching valid APs
 INFO probe_rs::session: Halting core 0...
 INFO probe_rs::session: Halting core 1...
 INFO probe_rs::session: Clearing breakpoints for core 0
 INFO probe_rs::session: Clearing breakpoints for core 1
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x10000000, virtual address: 0x10000000, flags: 0x5
 INFO probe_rs::flashing::download: Matching section: ".boot2"
 INFO probe_rs::flashing::download: Matching section: ".text"
 INFO probe_rs::flashing::download: Matching section: ".rodata"
 INFO probe_rs::flashing::download: Matching section: ".binary_info"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x100018d4, virtual address: 0x200000c0, flags: 0x7
 INFO probe_rs::flashing::download: Matching section: ".data"
 INFO probe_rs::flashing::loader: Found 2 loadable sections:
 INFO probe_rs::flashing::loader:     Multiple sections at 0x10000000 (6356 bytes)
 INFO probe_rs::flashing::loader:     .data at 0x100018D4 (296 bytes)
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20042000, cores: ["core0", "core1"], access: None }                            INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20042000, cores: ["core0", "core1"], access: None }                                INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack                                                                                      INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002378                                                                                                                               INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: None, range: 20000000..20042000, cores: ["core0", "core1"], access: None }
 INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: None, range: 20000000..20042000, cores: ["core0", "core1"], access: None }
 INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack
 INFO probe_rs::flashing::flash_algorithm: Stack top: 0x20002378
 INFO probe_rs::flashing::flasher: Erasing sector at address 0x10000000
 INFO probe_rs::flashing::flasher: Done erasing sector. Result is 0. This took 93.10532ms
 INFO probe_rs::flashing::flasher: Erasing sector at address 0x10001000
 INFO probe_rs::flashing::flasher: Done erasing sector. Result is 0. This took 12.22445ms
 INFO probe_rs::flashing::flasher: Took 100.857778ms to download 4096 byte page into ram
 INFO probe_rs::flashing::flasher: Took 97.723914ms to download 4096 byte page into ram
 INFO session_drop: probe_rs::session: Core 0 already halted                                                                                                                                   INFO session_drop: probe_rs::session: Halting core 1...                                                                                                                                       INFO session_drop: probe_rs::session: Clearing breakpoints for core 0                                                                                                                         INFO session_drop: probe_rs::session: Clearing breakpoints for core 1                                                                                                                         INFO probe_rs::architecture::arm::sequences: Powering down debug port Multidrop(11002927)                                                                                                     INFO probe_rs::architecture::arm::sequences: Powering down debug port Multidrop(1002927)                                                                                                     Error: The flashing procedure failed for 'blink.elf'.

Caused by:
    Flash content verification failed.
```

### Operating System

Linux (Debian)

bugadani, 2025-02-20:
Likely a duplicate of https://github.com/probe-rs/probe-rs/issues/2789

tti0, 2025-04-17:
It looks like this is fixed by https://github.com/probe-rs/probe-rs/pull/3156, but I will confirm locally and update the issue here.

tti0, 2025-04-17:
Indeed it is. Closing this issue. Thanks very much @bugadani!

closed 2025-04-17 by tti0