probe-rs/probe-rs #3114: No access port detected NUCLEO-G474RE / STM32G474RE

bene42, 2025-02-21:
### Describe the bug

Hi

I am unable to connect to STM32G474RE with the integrated ST-link V3 on a NUCLEO-G474RE eval board.
When running probe-rs info i get this:

```
> probe-rs.exe info
Probing target via JTAG

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
ARM Chip with debug port Default:
No access ports found on this chip.
Debug Port: DPv1, DP Designer: ARM Ltd

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

I also tried:
```
> probe-rs.exe info --chip STM32G474RETx --protocol SWD --log-file log.txt
Probing target via SWD

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
ARM Chip with debug port Default:
No access ports found on this chip.
Debug Port: DPv1, DP Designer: ARM Ltd

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

```

The log [log.txt](https://github.com/user-attachments/files/18902108/log.txt) is attached,

The probe is recognized:
```
> probe-rs.exe list
The following debug probes were found:
[0]: STLink V3 -- 0483:374e:0024003C3331510633323639 (ST-LINK)
```

STLink v3 firmware version:  V3J16

The probe works fine with openocd or from ST32CUBEIDE.

### To Reproduce

1. connect the NUCLEO-G474RE board to a PC
2. run probe-rs

### Expected behavior

The STM32G474RE is recognized by probe-rs.

### Stacktrace

```shell

```

### Operating System

Windows

### Additional context

Also tried this on Debian Trixie: No luck - Same result.

Yatekii, 2025-02-22:
Does flashing/running work? the `info` command is not indicative of probe-rs working.

bene42, 2025-02-22:
No, flashing does not work.

bene42, 2025-02-23:
```
> probe-rs run --chip STM32G474RETx --protocol SWD --log-file log.txt .\target\thumbv7em-none-eabi\release\blinky
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error which is specific to the debug probe in use occurred.
    3: Command failed with status JtagNoDeviceConnected.
```
The log [log.txt](https://github.com/user-attachments/files/18927433/log.txt) is attached.

Yatekii, 2025-02-23:
Did you try with `--connect-under-reset`? Also, do other tools work? This seems to strongly indicate that the chip is somehow not connected (which is more unlikely on devkits than other things)

bene42, 2025-02-23:
The Board is running some tests for at least 6 hours ATM. Will try `--connect-under-reset` right after that.
 As I mentioned in the report: "The probe works fine with openocd or from ST32CUBEIDE."
I can flash and debug the board with gdb/openocd and ST32CUBEIDE.

Yatekii, 2025-02-23:
> As I mentioned in the report

Sorry I missed that bit when checking on my phone :(

bene42, 2025-02-23:
> > As I mentioned in the report
> 
> Sorry I missed that bit when checking on my phone :(

No problem at all.

And I have good news! Flashing take ages, but work.
So `--connect-under-reset`  did the job.

For the record:
```
> probe-rs run --chip STM32G474RETx --protocol SWD --log-file log.txt --connect-under-reset .\target\thumbv7em-none-eabi\release\blinky
  Erasing ✔ 100% [####################]  18.00 KiB @  32.41 KiB/s (took 1s)
  Programming ✔ 100% [####################]  18.00 KiB @      246 B/s (took 75s)
     Finished in 75.35s
0.000000 TRACE BDCR ok: 00008200
└─ embassy_stm32::rcc::bd::{impl#3}::init @ H:\kicad\dcclicht\software\rust\embassy\embassy-stm32\src\rcc\bd.rs:216
0.000000 DEBUG rcc: Clocks { hclk1: MaybeHertz(16000000), hclk2: MaybeHertz(16000000), hclk3: MaybeHertz(16000000), hse: MaybeHertz(0), hsi48: MaybeHertz(48000000), pclk1: MaybeHertz(16000000), pclk1_tim: MaybeHertz(16000000), pclk2: MaybeHertz(16000000), pclk2_tim: MaybeHertz(16000000), pll1_p: MaybeHertz(0), pll1_q: MaybeHertz(0), rtc: MaybeHertz(32000), sys: MaybeHertz(16000000) }
└─ embassy_stm32::rcc::set_freqs @ H:\kicad\dcclicht\software\rust\embassy\embassy-stm32\src\rcc\mod.rs:71
0.000000 INFO  Hello World!
└─ blinky::____embassy_main_task::{async_fn#0} @ src\bin\blinky.rs:13
0.000091 INFO  high
└─ blinky::____embassy_main_task::{async_fn#0} @ src\bin\blinky.rs:18
0.300201 INFO  low
...
```

**Thanks** for helping out!

closed 2025-02-23 by bene42

Yatekii, 2025-02-23:
This is definitely not a normal speed :/ Could you try and turn off the log file? Also can you check the logs what speed the probe uses? I think the default on ST-Links is really low.

bene42, 2025-02-25:
`--log-file log.txt` was the culprit:

With log:
``´
> probe-rs run --chip STM32G474RETx --protocol SWD --log-file log.txt .\target\thumbv7em-none-eabi\release\blinky
      Erasing ✔ 100% [####################]  18.00 KiB @  31.76 KiB/s (took 1s)
  Programming ✔ 100% [####################]  18.00 KiB @      341 B/s (took 54s)
     Finished in 54.57s
0.000000 TRACE BDCR ok: 00008200
``´

Without log:
```
> probe-rs run --chip STM32G474RETx --protocol SWD .\target\thumbv7em-none-eabi\release\blinky
      Erasing ✔ 100% [####################]  18.00 KiB @  56.54 KiB/s (took 0s)
  Programming ✔ 100% [####################]  18.00 KiB @  40.74 KiB/s (took 0s)
     Finished in 0.76s
0.000000 TRACE BDCR ok: 00008200
```