probe-rs/probe-rs #3117: Error when programming the OTP bytes on a STM32, "No core can access the NVM region"

Yatekii, 2025-02-22:
### Discussed in https://github.com/probe-rs/probe-rs/discussions/2864

<div type='discussions-op-text'>

<sup>Originally posted by **ataffanel** October 15, 2024</sup>
Hi!

I need to program the OTP bytes on a ```stm32f405rgt```. I am trying first with the probe-rs CLI before moving to doing it in my program. I am using the version `0.24.0 (git commit: 9b97265f)` of probe-rs. I am getting the error that No core can acces the NVM region.

In my tries so far I get the:
```
❯ probe-rs download --chip stm32f405rgtx --binary-format bin --base-address 0x1fff7810 --log-file probe-rs.log myotp.bin 
Error: The flashing procedure failed for 'myotp.bin'.

Caused by:
    No core can access the NVM region NvmRegion { name: None, range: 536836096..536836624, cores: [], is_alias: false, access: Some(MemoryAccess { read: false, write: false, execute: false, boot: false }) }.
```

It does seem that probe-rs has knowledge of the OTP area though:
```
❯ probe-rs chip info stm32f405rgtx
stm32f405rgtx
Cores (1):
    - main (Armv7em)
NVM: 0x08000000..0x08100000 (1.0 MiB)
RAM: 0x10000000..0x10010000 (64.0 kiB)
RAM: 0x20000000..0x2001c000 (112.0 kiB)
RAM: 0x2001c000..0x20020000 (16.0 kiB)
NVM: 0x1fffc000..0x1fffc004 (4 B)
NVM: 0x1fff7800..0x1fff7a10 (528 B)
```

Is programming the OTP bytes possible with probe-rs, if so, how can I achieve it?</div>

Yatekii, 2025-02-22:
I am a bit confused because in the target definitions I see the flash algo to program the OTP ranges, but I do not see the OTP ranges themselves in the target definition.
Most likely the target definition back then was broken in a sense that the OTP region did not have main assigned as a core.

@bugadani @Dirbaio does stm32-data contain the OTP regs? I'd assume so, but then I am not sure how they don't land in the generated data. Maybe the core is still not assigned properly and they get ignored if they have no core?

bugadani, 2025-02-22:
IIRC since https://github.com/probe-rs/probe-rs/pull/2654 it doesn't matter if the memory region is defined. If there is an algorithm targeting it, probe-rs creates the memory region.

Yatekii, 2025-02-22:
Ah true. Maybe we do not assign the core properly to that region when we pull it from thin air?