probe-rs/probe-rs #3130: No defmt output from RTT after #3049

chrysn, 2025-02-26:
### Describe the bug

Since #3049, debug output that previously worked through RTT and defmt stopped showing anything.

The binaries I'm testing this with are build from Ariel OS, and I'm not 100% sure we don't do anything unexpected in there, but AIU that's just going through rtt-target.

### To Reproduce

1. Checkout https://github.com/ariel-os/ariel-os/
2. `laze -C examples/random/ build -b particle-xenon -D LOG=trace run`

  (that's not a minimal reproducer, but what I used for the logs)

4. That builds a binary and flashes it using probe-run, shown as `probe-rs run --protocol=swd --chip nrf52840_xxAA --preverify ../../build/bin/particle-xenon/cargo/thumbv7em-none-eabi/release/example-random`

I used this with a black-magic-probe; not sure if that makes any difference.

### Expected behavior

With probe-rs version b225227a5ecb737e708864f1b4a3bb1ae2a70b83 (before #3049), this prints:

```
     Running `probe-rs run --protocol=swd --chip nrf52840_xxAA --preverify ../../build/bin/particle-xenon/cargo/thumbv7em-none-eabi/release/example-random`
     Finished in 0.26s
DEBUG ariel_os_rt::startup()
DEBUG particle_xenon::init()
DEBUG nrf52::init()
DEBUG ariel-os-embassy::init(): using interrupt mode executor
DEBUG ariel-os-embassy::init_task()
DEBUG ariel-os-embassy::init_task() done
INFO  The random value of this round is 4.
INFO  The random value of this round is 1.
INFO  The random value of this round is 4.
INFO  The random value of this round is 2.
INFO  The random value of this round is 4.
INFO  The random value of this round is 6.
INFO  The random value of this round is 4.
INFO  The random value of this round is 4.
INFO  The random value of this round is 6.
INFO  The random value of this round is 4.
```

Instead, output stops at:

```
     Finished in 0.26s
```

### Stacktrace

```shell

```

### Operating System

Linux (Debian)

### Additional context

I've attached logs for comparison:
[good.log](https://github.com/user-attachments/files/18984404/good.log) 
[bad.log](https://github.com/user-attachments/files/18984409/bad.log)

(The bad log is slightly longer because I Ctrl-C'd it only I was sure it had enough time, whereas I know when all output was through for the good one.)

bugadani, 2025-02-27:
> but with the caveat that the text quickly scrolls out of the screen.

That was supposed to be fixed by https://github.com/probe-rs/probe-rs/pull/3108

bugadani, 2025-03-01:
Oh, I see what went wrong. I remembered incorrectly when I assumed `run` just printed channel 0 - in #3049 apparently the old code collected every channel into a single output stream, ignoring the source channel. Changing the current implementation to do the same would be trivial (we'd just need to remove [this check](https://github.com/probe-rs/probe-rs/blob/3c467b66c8b8121077921c635df69ceee04af2d8/probe-rs-tools/src/bin/probe-rs/util/cli.rs#L502)). This has a lot of assumptions attached (if the .elf contains defmt data, every channel would be treated as defmt, incorrectly). Even if a chanel is used to stream binary data, we would try to either defmt-decode it, or treat it as a string, which is not ideal.

Instead I want probe-rs to be a bit smarter, but any heuristic will likely fail on some edge case. At first, though, the following should be plenty good enough:

- If the channel name is "defmt" and there is defmt data in the .elf, decode as defmt
- Otherwise just use `String::from_utf8_lossy`.
- If there are multiple channels, print the channel name next to the message.

To get this to work, I need to find a way to communicate the RTT channel names through the RPC interface. The lazy solution (that will likely be implemented) would be to add a `MonitorEvent::RttConnected { channels }` message. #3134 is a tiny step in this direction.

More to come.

chrysn, 2025-03-01:
Thanks for looking into this.

From the perspective of my use case, I'm fine if the heurstics change (and even proposed altering our channel sequence as a quick fix in https://github.com/ariel-os/ariel-os/pull/878 -- after all, the sequence doesn't really matter internally).

As for what to do with multiple output streams, [cargo-embed has examples where they also transfer binary data](https://probe.rs/docs/tools/cargo-embed/#external-front-end-via-tcp-socket). The "show all channels as from_utf8_lossy with prefix" strategy would look bad there, but that may be an acceptable corner case.

bugadani, 2025-03-02:
Could you try https://github.com/probe-rs/probe-rs/pull/3140 please?

PR #3140 (2025-03-05): Print data from all defmt channels
Closes #3130