probe-rs/probe-rs #3131: S32K142 core locked after first program.

liamkinne, 2025-02-27:
### Describe the bug

I am trying to program S32K1 devices with an NXP MCU-Link programmer.

Here is what I captured from the initial flash:
```
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.12s
     Running `probe-rs run --chip S32K142UAxxxLLx --protocol swd --connect-under-reset /Users/liam/umi/pdm/target/thumbv7em-none-eabihf/debug/pdm-36-k1`
      Erasing ✔ 100% [####################]   8.00 KiB @  31.47 KiB/s (took 0s)
  Programming ✔ 100% [####################]   8.00 KiB @  22.71 KiB/s (took 0s)             Finished in 0.61s
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    1: Failed to read register DRW at address 0xd0c
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.
Error: A timeout occurred.
```

Here is what `probe-rs info` now shows after the device has become inaccessible:

```
Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
├── Memory AP is not accessible, DeviceEn bit not set
└── 1 Unknown AP (Designer: Freescale (Motorola), Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x0)
```

### To Reproduce

1. Attach debugger to SWD port.
2. Power on device.
3. `probe-rs run --chip S32K142UAxxxLLx --protocol swd --connect-under-reset`

### Expected behavior

Even if RTT fails to attach, the device should be able to be programmed again.

### Stacktrace

```shell
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    2: Failed to read register DRW at address 0xd0c
    3: An error occurred in the communication with an access port or debug port.
    4: Target device responded with a FAULT response to the request.
```

### Operating System

macOS

### Additional context

[debug.txt](https://github.com/user-attachments/files/19004345/debug.txt)

xobs, 2025-02-28:
I'm not able to see `debug.txt`, but I wonder if you didn't overwrite the Flash Security Byte. That's the byte at offset `0x40C` that determines what you can do, and if the last two bits are anything other than `0b10` then it won't be available under SWD anymore.

I don't see anything in `probe-rs` that explicitly patches those bytes. It may be worth it to add that for Kinetis. For example, [pyOCD](https://github.com/pyocd/pyOCD/blob/90ee63a57a73ccca6da10b4ad5e3ebd5a6775e58/pyocd/target/family/flash_kinetis.py#L40), [openocd](https://github.com/openocd-org/openocd/blob/91c11ea469c7613d2b474149c48457dcc082ebed/src/flash/nor/kinetis.c#L3321-L3325), and [Blackmagic](https://github.com/blackmagic-debug/blackmagic/blob/3148437ed5b1d966616713c5362927c9e8dbcd96/src/target/kinetis.c#L477-L481) disallow writes to the FCF region if a `--dangerous`-style flag hasn't been specified.

What's the value at address `0x40C` in your program?

xobs, 2025-03-01:
You should still be able to issue a `mass_erase` on the chip by talking to the MDM. I'm not sure how that's done in `probe_rs`, but in Blackmagic it's handled by the programming routine talking knowing how to talk to the MDM-AP directly, and on openocd it's through the [mass_erase](https://github.com/openocd-org/openocd/blob/91c11ea469c7613d2b474149c48457dcc082ebed/src/flash/nor/kinetis.c#L3373-L3379) command.

As long as it's still responding over SWD, you can run those commands.

liamkinne, 2025-03-01:
I have figured out a linker script that does something similar to what the S32 Design Studio tooling does, putting the flash config in the right address.

```
MEMORY
{
  FLASH : ORIGIN = 0x00000000, LENGTH = 256K
  RAM : ORIGIN = 0x1FFFC000, LENGTH = 16K
  RAM2 : ORIGIN = 0x20000000, LENGTH = 12K
}

SECTIONS
{
  .interrupts :
  {
    __VECTOR_TABLE = .;
    . = ALIGN(4);
    KEEP(*(.isr_vector))
    . = ALIGN(4);
  } > FLASH

  .flash_config 0x400 :
  {
    . = ALIGN(4);
    KEEP(*(.FlashConfig))
    . = ALIGN(4);
  } > FLASH
}

_stext = ORIGIN(FLASH) + 0x410;
```

Obviously still probe-rs should protect devices from being bricked by default as I can imagine it being quite easy to flash the wrong file or use the wrong linker script.

Would be best way to go about this be to write a custom [flash algorithm](https://github.com/probe-rs/flash-algorithm), or is this something that should be done at the probe-rs lib level?

liamkinne, 2025-03-09:
Just bricked another chip trying to launch into debugging. Down to two of my original five samples. If this continues I'm just going to abandon using this mcu and pick something a little less temperamental.

p3zhy, 2025-03-10:
Hey @liamkinne , 

I saw you're having trouble with debugging the S32K142 using probe-rs. I ran into similar issues before and found that using the **Cortex-Debug** extension for VS Code, along with J-Link, worked more reliably for me.  

I've put together a template in my GitHub repo that you might find useful. It includes a **launch.json** setup for **Cortex-Debug** with J-Link, targeting an S32K144 (which should be similar to your S32K142). Here's the config I use:  

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "device": "S32K144",
            "interface": "jtag",
            "executable": "./target/thumbv7em-none-eabihf/debug/s32k-quickstart",
            "runToEntryPoint": "true",
            "svdFile": "./S32K144.svd",
            "preLaunchTask": "cargo build"
        }
    ]
}

I still run into some quirks with debugging, but at least I can reliably connect and step through the code. If you're open to trying a different approach, you can check out my repo here: [s32k-quickstart](https://github.com/p3zhy/s32k-quickstart)
Let me know if you need help setting it up!

Hope this helps.