probe-rs/probe-rs #3146: Debugging stm32wb55 using JLink / STLink does not work

ckrenslehner, 2025-03-06:
### Describe the bug

See https://github.com/probe-rs/probe-rs/discussions/3145

I asked about this in the Rust Embedded Matrix Chat and got the feedback, that there is something wrong. 
I tested version 0.25 and everythink is working as expected. Version 0.26 does not work anymore. I am trying to pin down the exact commit where things start to break. 

Thanks for the quick response in the chat! :-)

### Stacktrace

```shell

```

### Operating System

None

ckrenslehner, 2025-03-06:
Commit c1de4a9 does not work

ckrenslehner, 2025-03-06:
24d70da works!

So the bug is hidden somewhere in between

ckrenslehner, 2025-03-07:
Hi, I pinned down the commit where things start to break.

[image]

The checked out HEAD ( 2bb002b ) does work. 
07c64f7 does not work anymore.

noppej, 2025-03-07:
Good sleuthing!   You’ve probably already thought about this, but can you try downgrading the `itertools` dependency that was changed in that commit, using master branch?

[image]

ckrenslehner, 2025-03-07:
I will try tomorrow

bugadani, 2025-03-08:
Can you please try https://github.com/probe-rs/probe-rs/pull/3155 ?

ckrenslehner, 2025-03-08:
I will try now, thanks :-)

ckrenslehner, 2025-03-08:
I think there is some progress, but it still does not work, unfortunately.

There is more output from the `info` command.

```
probe-rs on  jlink is 📦 v0.27.0 via 🦀
❯ : probe-rs --version
probe-rs 0.27.0 (git commit: v0.27.0-47-g12169c6b)

probe-rs on  jlink is 📦 v0.27.0 via 🦀
❯ : probe-rs info
Probing target via JTAG
-----------------------

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Failed to identify target using protocol JTAG: An error with the usage of the probe occurred

Caused by:
    0: An error occurred handling the JTAG scan chain.
    1: Invalid IR scan chain

Probing target via SWD
----------------------

 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR1 has invalid preamble (expected 0x0, got 0x1)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR2 has invalid preamble (expected 0x5, got 0xb1)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0040000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0040000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
ARM Chip with debug port Default:

Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4950, Revision: 0x0, Instance: 0x00
├── V1(0) MemoryAP
│   └── 0 MemoryAP (AmbaAhb3)
│       ├── 0xe00ff000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe0001000 Generic
│       ├── 0xe0000000 Peripheral test block
│       ├── 0xe0040000 Generic
│       ├── 0xe0041000 Cortex-M4 ETM   (Coresight Component)
│       └── 0xe0043000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
└── V1(1) MemoryAP
    └── 1 MemoryAP (AmbaAhb3)
        └── 0xf0000000 Generic
```

Still I get the following errors:

[image]

```
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 56411
probe-rs-debug: Starting debug session from: 127.0.0.1:56413
FLASHING: Starting write of "C:\\Users\\ChristianKrenslehner\\Documents\\Software\\firmware-stm32-embassy\\./target/thumbv7em-none-eabihf/debug/firmware-stm32-embassy" to device memory
FLASHING: Completed write of "C:\\Users\\ChristianKrenslehner\\Documents\\Software\\firmware-stm32-embassy\\./target/thumbv7em-none-eabihf/debug/firmware-stm32-embassy" to device memory
ERROR: probe-rs-debug: "Debug Adapter terminated unexpectedly with an error: Other(Cannot continue unless one target core configuration is defined.)\n"
 WARN probe_rs::session:835: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: A USB transport error occurred.
    3: timed out
 WARN probe_rs::session:847: Failed to deconfigure device during shutdown: Arm(Probe(ProbeSpecific(BoxedProbeError(Usb(Kind(TimedOut))))))
probe-rs-debug: Session ended: Cannot continue unless one target core configuration is defined.
probe-rs-debug: DAP Protocol server exiting

probe-rs-debug: Closing probe-rs debug session
```

```
probe-rs on  jlink is 📦 v0.27.0 via 🦀 took 10s
❯ : probe-rs list
The following debug probes were found:
[0]: J-Link -- 1366:1024:000601006224 (J-Link)
```

ckrenslehner, 2025-03-08:
Programming with STLinkV3 does not work anymore

```
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 56532
probe-rs-debug: Starting debug session from: 127.0.0.1:56534
An error with the usage of the probe occurred
	"An error which is specific to the debug probe in use occurred."
		"Command failed with status SwdDpFault."
```

ckrenslehner, 2025-03-08:
IDK if this helps:
If I use the JLinkConfigurator and set the JLink to CMSIS-DAP mode, things work again.

ckrenslehner, 2025-03-08:
This is probably a separate issue, but the "Restart" action in VsCode does not work as well. I need to click the "Stop" button and then start debugging again. I don't know if this should already work TBH so forgive me, if this is not yet implemented.

bugadani, 2025-03-08:
> IDK if this helps:
If I use the JLinkConfigurator and set the JLink to CMSIS-DAP mode, things work again.

I guess that's a good enough solution for you, but unfortunately it doesn't help us identify what's wrong.

ckrenslehner, 2025-03-09:
I tested again using the JLink only (no STLinkV3).

Commit 712ecf7be7938ae35a3a8eaa07959df9ce140e01 breaks the JLink debugging.
Commit 2bb002b68e6ba4e3d2a15837c0884639d3a196cf works fine.

Here some additional output:

WORKING
```
D:\Software\probe-rs> probe-rs --version                                                                                                   03/09/25 11:50:01 AMprobe-rs 0.25.0 (git commit: v0.25.0-16-g2bb002b6)
D:\Software\probe-rs> probe-rs info                                                                                                        03/09/25 11:50:06 AM
Probing target via JTAG

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xf0000000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4950, Revision: 0x0, Instance: 0x00
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: STMicroelectronics
│       ├── Cortex-M4 SCS   (Generic IP component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M4
│       │       └── REVISION: 1
│       ├── Cortex-M3 DWT   (Generic IP component)
│       ├── Cortex-M3 FBP   (Generic IP component)
│       ├── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M4 TPIU  (Coresight Component)
│       ├── Cortex-M4 ETM   (Coresight Component)
│       └── Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
└── 1 MemoryAP (AmbaAhb3)
    └── Generic

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```
```
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 36012
probe-rs-debug: Starting debug session from: 127.0.0.1:36015
FLASHING: Starting write of "D:\\Software\\firmware-stm32-embassy\\./target/thumbv7em-none-eabihf/debug/firmware-stm32-embassy" to device memory
FLASHING: Completed write of "D:\\Software\\firmware-stm32-embassy\\./target/thumbv7em-none-eabihf/debug/firmware-stm32-embassy" to device memory
probe-rs-debug: RTT Window reused, and ready to receive RTT data on channel 0
```
What is interesting is this: 
As soon, as I hit the "Stop" Button, I also get the weird error message with the missing core configuration.

[image]
[image]

There is also an audible USB disconnect and reconnect sound on my Windows PC. But at least debugging works.

NOT WORKING
```
D:\Software\probe-rs> probe-rs --version                                                                                                   03/09/25 11:53:04 AMprobe-rs 0.25.0 (git commit: v0.25.0-17-g712ecf7b)
D:\Software\probe-rs> probe-rs info                                                                                                        03/09/25 11:53:11 AM
Probing target via JTAG

ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD
[... 36 lines cut ...]
Caused by:
    0: The debug probe encountered an error.
    1: An error which is specific to the debug probe in use occurred.
    2: A USB transport error occurred.
    3: timed out
 WARN probe_rs::session:844: Failed to deconfigure device during shutdown: Arm(Probe(ProbeSpecific(BoxedProbeError(Usb(Kind(TimedOut))))))
probe-rs-debug: Session ended: Cannot continue unless one target core configuration is defined.
probe-rs-debug: DAP Protocol server exiting

probe-rs-debug: Closing probe-rs debug session
```

The `probe-rs info` commands seems to work the same with both commits, however there seems to be a USB timeout after going to 712ecf7be7938ae35a3a8eaa07959df9ce140e01 

The information of the comments above might only be related to the STLinkV3 issue of stopping but not showing the yellow bar above the line. I will double-check this again. Sorry for the confusion.

My testing procedure is as follows:
- Add breakpoint
- Start debuggin
- Hit continue 
- Wait for stopping at breakpoint

So as conclusion:
- 2bb002b68e6ba4e3d2a15837c0884639d3a196cf  works. As soon as I hit stop, I get the error message and a USB reconnect sound.
- 712ecf7be7938ae35a3a8eaa07959df9ce140e01  does not work

ckrenslehner, 2025-03-09:
> This is probably a separate issue, but the "Restart" action in VsCode does not work as well. I need to click the "Stop" button and then start debugging again. I don't know if this should already work TBH so forgive me, if this is not yet implemented.

I have an update to this as well. 
If I use version 0.24 also restarting works. This might be related with the error which is thrown when clicking the stop button. Maybe the root cause of the USB timeout was introduced between versions 0.24 and 0.25 after all. I will keep looking. 👁

bugadani, 2025-03-09:
Your breakpoint issue may have been fixed by https://github.com/probe-rs/probe-rs/pull/3137 already

ckrenslehner, 2025-03-09:
The restarting seems to break with the commit bd68439291c439e289b4544fcdba0c3285f187ff. If I use the commit right below (496b918ba458ced114cc7227d6752a9d68bc12a2) everything works.

Looking at the commit history, this does not really make any sense?

ckrenslehner, 2025-03-09:
> Your breakpoint issue may have been fixed by [#3137](https://github.com/probe-rs/probe-rs/pull/3137) already

The "yellow bar" issue?

ckrenslehner, 2025-03-09:
> The restarting seems to break with the commit [bd68439](https://github.com/probe-rs/probe-rs/commit/bd68439291c439e289b4544fcdba0c3285f187ff). If I use the commit right below ([496b918](https://github.com/probe-rs/probe-rs/commit/496b918ba458ced114cc7227d6752a9d68bc12a2)) everything works.
> 
> Looking at the commit history, this does not really make any sense?

This is probably not correct. Further testing also showed issues with (https://github.com/probe-rs/probe-rs/commit/496b918ba458ced114cc7227d6752a9d68bc12a2). When I stick to 0.24 installed via `cargo binstall probe-rs-tools --version 0.24` everything works flawlessly. Restarting and all of it.

ckrenslehner, 2025-04-02:
Hi again!
I added a minimal firmware to reproduce the debugging issue

https://github.com/ckrenslehner/embedded-test-issue/tree/feat/probe-rs-jlink-issue

The test procedure and the results are described in the README

I can debug with v0.25 but not with v0.27. Unfortunately, it's harder than expected to get an exact commit, where things start to break apart.

Maybe someone can also reproduce this issue