probe-rs/probe-rs #3158: Loosing messages

Murmele, 2025-03-09:
### Describe the bug

When printing without any waittime with defmt, I am loosing messages

### To Reproduce

1. Embedded target
2. Add the following code to the Main:
```rust
let mut counter = 0;
loop {
    error!("Counter: {}", counter);
    counter += 1;
}
```

### Expected behavior

No counter values get lost

### Stacktrace

```shell

```

### Operating System

None

### Additional context

Sometimes I am loosing values
```
111.882415 Counter: 324032
111.882476 Counter: 324033
111.882537 Counter: 324034
111.882629 Counter: 324036
111.882690 Counter: 324037
111.882751 Counter: 324038
```

Current master I compiled: 808a6c6736d516688cc72864352946a29b419141

bugadani, 2025-03-09:
Add some details, please:
- which command do you run?
- what MCU do you use?
- what channel settings do you specify?

Murmele, 2025-03-10:
I have a NRF52840 mcu and used: `probe-rs run --log-format "{t} {s}" --chip nRF52840_xxAA <binary path>` to flash the file and receiving debug messages. 

How do I get the channel settings?