probe-rs/probe-rs #3174: Unable to send RTT data to device using down channel socket

Robbie-W1, 2025-03-15:
### Describe the bug

Receiving data on a socket from an up channel works fine, but sending data from the host to the target via aa socket does not work

### To Reproduce

1. Set up probe-rs with a down channel configured with a socket
2. Flash the firmware and connect to the socket
3. Try and send data over the socket - the device does not receive it

### Expected behavior

I would expect an application which can read rtt data via a socket to also be able to write it

### Stacktrace

```shell

```

### Operating System

Linux (Ubuntu)

### Additional context

Using an RP2040 with picoprobe, running Ubuntu via WSL2

bugadani, 2025-03-15:
This issue is related to cargo-embed, right? Probe-rs has otherwise no support for down channels at all.

Robbie-W1, 2025-03-15:
@bugadani yes this is for cargo-embed. I've looked at probe-rs-tools/src/bin/probe-rs/cmd/cargo_embed/config/mod.rs and it seems there is no support for sockets on down channels so I guess this is a feature request rather than a bug. 

Is there another way to send data on a down channel programmatically? The pr0be-rs documentation seems to be a bit lacking here.

bugadani, 2025-03-16:
Not currently, this use case was not really considered I think.