probe-rs/probe-rs #3180: Feature Request: Option for probe-rs to read configuration from Embed.toml

Hecatron, 2025-03-17:
### Is your feature request related to a problem? Please describe.

Hi,
I've noticed that cargo embed has it's own configuration file typically `Embed.toml`
In the case of probe-rs this doesn't have a configuration file but relies typically on command line arguments
which can be passed in via the vscode extension via `launch.json`

Since there's a degree of overlap / duplication I was wondering if it might be an idea to have a command line argument for probe-rs to try and read the same `Embed.toml` configuration to pull some settings such as the chipset, if to reset, log_level that sort of thing.
This way you'd just have a single Embed.toml file at the root of the project, and `launch.json` could be made to be smaller by just telling probe-rs to pull its settings from the same Embed.toml file

This might mean adding some probe-rs specific settings to the toml file though if these don't normally apply to cargo embed.

bugadani, 2025-03-17:
This won't happen. probe-rs has its own configuration file, we'll add things to that, but probe-rs has no business reading Embed.toml.

bugadani, 2025-03-17:
It's mostly related to the upcoming server funcitonality, and it's mostly expected as a user-specific global config file, but that can be changed I guess. But we already have a way to configure probe-rs from the CLI, and from environment variables anyway.

To be quite honest, if you're already using cargo embed, I'm just not sure why you'd want to use the probe-rs CLI.

Hecatron, 2025-03-17:
I was trying to see it from a context of someone who is writing a publicly visible project, but doesn't know how the end user is going to debug or which tool they will use (since some users may prefer some tools over others, or different debug hardware).

one feature I like is that cargo embed has for example is the profiles feature where you can just switch the settings around by just using a single cli flag. I just figured it might help cut down on all the settings within launch.json if the majority could be broken out into a separate toml file

If it's not a good idea then please feel free to close though