probe-rs/probe-rs #3187: Probe not found during remote run

dvdsk, 2025-03-23:
### Describe the bug

While trying to deploy and run on a remote raspberry pi I get the error: `no connected probes were found`

### To Reproduce

1. build probe-rs from git, specifically commit g91fffa8 with `--features remote`
2. start a probe-rs server with `probe-rs serve`
3. list the probes connected to the server  `probe-rs list --host 'ws://192.168.1.43:3000' --token secret_token_redacted`
4. try to run a binary using `probe-rs run --host 'ws://192.168.1.43:3000' --token secret_token_redacted /<long_path>/target/thumbv7em-none-eabihf/debug/sensor-node`

## Alternative
4. try to run a binary using `probe-rs run --host 'ws://192.168.1.43:3000' --token secret_token_redacted /<long_path>/target/thumbv7em-none-eabihf/debug/sensor-node --probe c251:f001:202108120001"`

### Expected behavior

Similar behavior to running locally.

### Stacktrace

```shell

```

### Operating System

Other Linux Distribution

### Additional context

output of `RUST_LOG=debug,nusb::platform::linux_usbfs::enumeration=info probe-rs serve` 
it is identical whether `probe-rs run` is done with or without `--probe c251:f001:202108120001`
```
INFO probe_rs::cmd::serve: listening on 0.0.0.0:3000
INFO probe_rs::cmd::serve: User david connected
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
DEBUG nusb::platform::linux_usbfs::device: Opening usbfs device /dev/bus/usb/001/004
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open c251:f001 in CMSIS-DAP v1 mode
INFO probe_rs::cmd::serve: Client disconnected
```

output of `RUST_LOG=debug probe-rs run --host 'ws://192.168.1.43:3000' --token secret_token_redacted --probe c251:f001:202108120001 /<long_path>/target/thumbv7em-none-eabihf/debug/large-bedroom-sensor-node` 
```
DEBUG tungstenite::handshake::client: Client handshake done.
DEBUG probe_rs::cmd::run: No embedded-test in ELF file. Running as normal
DEBUG postcard_rpc::host_client::util: Handled message via map
DEBUG postcard_rpc::host_client::util: Handled message via map
Error: Probe not found
```

bugadani, 2025-03-23:
Thanks for reporting this, I'll see if I can find a CMSIS DAP probe and test this myself.

> Similar behavior to running locally.

So does this mean probe-rs can work with that probe when you try to use it on your raspberry in a local session?

dvdsk, 2025-03-23:
> So does this mean probe-rs can work with that probe when you try to use it on your raspberry in a local session?

oh damn, I did not test that. Now that I did I get the same error locally on the Pi. So apologies this bug report seems to be a false positive. I'll check further and let you know.

dvdsk, 2025-03-23:
Its an issue with running on the PI. I'll close this and open a new one if I can not figure it out 😅

closed 2025-03-23 by dvdsk