probe-rs/probe-rs #3225: Probe undetected regardless of udev rules

ofluffydev, 2025-04-13:
### Describe the bug

The ordering/filename issue isn't applied here, it's the same rule number right from the downloads.

I've restarted the pico a ton and tried everything the docs said, `picotool load -u -v -x -t elf` works fine and I was using it before but I need debugging since waveshare displays aren't the best. This probably has something to do with my OS being Arch linux but I can't tell since the tool just denies the existence of the "probe."

Heres the command:
```
cargo embed                                                                                                                [22:36:29]
      Config default
      Target /home/fluffy/Code/inky/target/thumbv8m.main-none-eabihf/debug/inky
       Error No connected probes were found.
```

### To Reproduce

Use probe-rs for RP235x chip on an arch linux machine

### Stacktrace

```shell

```

### Operating System

None

ofluffydev, 2025-04-13:
Fails to find the pico 2 w as root as well

ofluffydev, 2025-04-14:
> Which debug probe are you using? Note that the usb download mode on the
> rp2350 doesn't count as a debug probe.

defmt and rtt if that's what you're asking? It doesn't even detect the pico
for flashing though with any of the commands provided

ofluffydev, 2025-04-14:
> No, I mean what's the box between the pico and your PC

man.. there's supposed to be something else?.. Why flash and enter debug if
you can't just get debugging back from the same USB.. I was hoping for
something similar to the pi zero serial terminal over USB. Is there no way
to debug from just the plug on the pico straight to the computer?

bugadani, 2025-04-14:
You should be able to, yes. Read through the page I linked, especially the `Serial Wire Debug (SWD)` section.