probe-rs/probe-rs #3226: Incorrect docs i guess?

ofluffydev, 2025-04-13:
### Describe the bug

Default tutorial will lead you to this if you try to run `cargo embed` with your target set as rp235x

```
error: could not compile `inky1` (bin "inky1") due to 1 previous error
error: linking with `cc` failed: exit status: 1
  |
  = note:  "cc" "-m64" "/tmp/rustcbLCpIS/symbols.o" "<3 object files omitted>" "-Wl,--as-needed" "-Wl,-Bstatic" "/home/fluffy/Code/inky1/target/debug/deps/{librtt_target-e0ccecd9cb01ff1a.rlib,libcritical_section-b0795fc99eb80917.rlib,libportable_atomic-229f3c0a4af89f97.rlib,libufmt_write-0f6d9989081d8233.rlib,libmicrobit-fba608e1f21008c5.rlib,libmicrobit_common-d15f21825dfa0db3.rlib,libtiny_led_matrix-7e480e28453a5875.rlib,libnrf51_hal-8b7dbd9eda51d2f3.rlib,libnrf_hal_common-91a713efc9baa394.rlib,libembedded_io-cf3a495dd41d6b81.rlib,libfixed-d2a3e7381a13ea34.rlib,libtypenum-8dd2d9ed8e949aa6.rlib,libbytemuck-5393e4f51b211558.rlib,libhalf-7456cb8afddaf66d.rlib,libaz-552a7523d2366f66.rlib,librand_core-7ade737e98575d96.rlib,libcfg_if-640f4ab12ae8a3ab.rlib,libembedded_hal-05a01a246d9d0118.rlib,libnrf51_pac-7c7cc5003f05b5f2.rlib,libcortex_m_rt-6baf1b7397242f89.rlib,libcortex_m-6cbee47832654222.rlib,libembedded_hal-d100e75d3c7d79e5.rlib,libvoid-a7e5a317f75f5711.rlib,libnb-5dc64c8cc047ca9d.rlib,libnb-21a4aa96921a6ebb.rlib,libvolatile_register-480c07a974cb4f82.rlib,libvcell-26f90eab12ee968e.rlib,libbare_metal-9dd2bc2111a4d3a2.rlib}.rlib" "<sysroot>/lib/rustlib/x86_64-unknown-linux-gnu/lib/{librustc_std_workspace_core-*,libcore-*,libcompiler_builtins-*}.rlib" "-L" "/tmp/rustcbLCpIS/raw-dylibs" "-Wl,-Bdynamic" "-B<sysroot>/lib/rustlib/x86_64-unknown-linux-gnu/bin/gcc-ld" "-fuse-ld=lld" "-Wl,-znostart-stop-gc" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/fluffy/Code/inky1/target/debug/build/cortex-m-rt-dd3508a00a39e44b/out" "-L" "/home/fluffy/Code/inky1/target/debug/build/nrf51-hal-c9609a31b81b0741/out" "-L" "/home/fluffy/Code/inky1/target/debug/build/nrf51-pac-115f65c65e808d5b/out" "-L" "<sysroot>/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/fluffy/Code/inky1/target/debug/deps/inky1-c22a372b09a16d5b" "-Wl,--gc-sections" "-pie" "-Wl,-z,relro,-z,now" "-nodefaultlibs"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: undefined symbol: rust_eh_personality
          >>> referenced by core.8b7fb098a05ab0cb-cgu.0
          >>>               core-11d9a250f9da47d5.core.8b7fb098a05ab0cb-cgu.0.rcgu.o:(DW.ref.rust_eh_personality) in archive /home/fluffy/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcore-11d9a250f9da47d5.rlib
          
          rust-lld: error: undefined symbol: main
          >>> referenced by /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/../../../../lib/Scrt1.o:(_start)
          
          rust-lld: error: undefined symbol: __libc_start_main
          >>> referenced by /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/../../../../lib/Scrt1.o:(_start)
          collect2: error: ld returned 1 exit status
```

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2025-04-13:
Please point out exactly what docs you're following

ofluffydev, 2025-04-13:
tried everything on https://probe.rs/docs/overview/about-probe-rs/#probe-rs

bugadani, 2025-04-13:
And why do you think that is relevant for your project failing to compile?