probe-rs/probe-rs #3231: "No debug probes were found." on Mac OS X for Arduino Nano and for ESP8266

kjkoster, 2025-04-16:
I've connected two devices to my Mac separately: an Arduino Nano 33 IoT and an ESP8266 NodeMCU Amica. Neither are detected using rs-probe and it would be helpful if they were.

Pretty new to Rust embedded, so this may well be driver error. Guidance appreciated.

Installed using the download-script-into-`sh` trick:

```
$ probe-rs --version
probe-rs 0.27.0 (git commit: 7d59a31)
```

First, the ESP8266. This shows up as `/dev/cu.usbserial-0001` on my Mac.

```
$ probe-rs info
Error: No connected probes were found.
$ probe-rs list
No debug probes were found.
$ ls -l /dev/cu.usbserial-0001
crw-rw-rw-  1 root  wheel  0x9000005 16 Apr 07:50 /dev/cu.usbserial-0001
```

Then, I switched out the ESP for an Arduino Nano 33 IoT.

```
$ probe-rs info
Error: No connected probes were found.
$ probe-rs list
No debug probes were found.
$ ls -l /dev/cu.usbmodem1101
crw-rw-rw-  1 root  wheel  0x9000005 16 Apr 07:56 /dev/cu.usbmodem1101
```

closed 2025-04-16 by bugadani

kjkoster, 2025-04-16:
I guess I was confused because with the Micro:bit `probe-rs` does work to flash the devices. Do Micro:bits come with probes integrated? How come they do work?

bugadani, 2025-04-16:
The microbits (and STM devkits, and newer ESP32, and more), each have the debug probe integrated in some way. On microbit and STM32 there is an interface chip that implements the functionality. On newer ESP32 chips (S3, C6, H2, ...) the interface is built into the MCU itself.

kjkoster, 2025-04-16:
Very helpful of you to explain. Thank you!