probe-rs/probe-rs #3255: “No Table definition in DWARF info” even when RTT symbol is present (_SEGGER_RTT)

niekiran, 2025-04-25:
Hi, 
When I flash & run with `probe-rs run --chip STM32F303CC `.  the firmware is programmed successfully, but I immediately get this warning and i dont observe any debug logs coming out from my program
`WARN probe_rs::util::rtt: No `Table` definition in DWARF info; compile your program with `debug = 2` to enable location info.`

The ELF was built with `profile.dev.debug = 2.`
arm-none-eabi-nm shows both _SEGGER_RTT and the defmt_rtt::RTT_ENCODER symbols.
RTT/defmt output never appears

can anyone help to fix this issue? Thank you

minimal code 
```
use cortex_m_rt::entry;
use defmt_rtt as _;
use panic_probe as _;

#[entry] fn main() -> ! {
    defmt::info!("hello");
    loop {}
}
```
Cargo.toml 
```
[profile.dev]
debug = 2

# [... full dependency list omitted for brevity ...]
```

niekiran, 2025-04-25:
Hi , 
Thank you for quick reply , 
here is full code main.rs 
```
#![no_main]
#![no_std]

use cortex_m::peripheral::syst::SystClkSource;
use cortex_m_rt::entry;
use defmt_rtt as _;            
use panic_probe as _;          
use stm32f3::stm32f303; 
#[entry]
fn main() -> ! {
   
    let cp = cortex_m::Peripherals::take().unwrap();

    // --- SysTick setup ---------------------------------------------
   
    let mut syst = cp.SYST;
    syst.set_clock_source(SystClkSource::External);  
    syst.set_reload(1_000_000 - 1);                  // 1 Hz period
    syst.enable_counter();
    defmt::info!("Hello, world");

   
    loop {
        
        while !syst.has_wrapped() {}
        defmt::info!("Hello, world");
    }
}

fn monotonics_timestamp() -> u64 {
    0
}

defmt::timestamp!("{=u64}", monotonics_timestamp());
```
Cargo.toml 

```
[package]
name = "log-test"
version = "0.1.0"
edition = "2021"

[dependencies]
cortex-m       = { version = "0.7.7", features = ["critical-section-single-core"] }
cortex-m-rt    = "0.7.5"
defmt          = "1.0.1"
defmt-rtt      = "1.0.0"
panic-probe    = { version = "1.0.0", features = ["print-defmt"] }

[dependencies.stm32f3]
version = "0.15.1"
features = ["stm32f303", "rt"]

[profile.dev]
debug = 2
```

bugadani, 2025-04-25:
This is still not a complete project. Are you referencing the `defmt.x` linker script in your project, for example?

niekiran, 2025-04-25:
Hi Sorry, 
.cargo/config.toml file 

```
[build]
target = "thumbv7em-none-eabihf"

# Settings that apply only to that target triple
[target.thumbv7em-none-eabihf]
runner    = "probe-rs run --chip STM32F303CC"   # ← adjust part number
rustflags = [
    "-C", "link-arg=-Tlink.x",
  
]
```

memory.x 

```
/* Define memory regions */
MEMORY
{
    FLASH (rx)    : ORIGIN = 0x08000000, LENGTH = 256K
    RAM (rwx)     : ORIGIN = 0x20000000, LENGTH = 40K
  /*  EEPROM (rwx)  : ORIGIN = 0x08080000, LENGTH = 4K  */
    CCMRAM (rwx)  : ORIGIN = 0x10000000, LENGTH = 8K  
  /*  BATTRAM (rw)  : ORIGIN = 0x40024000, LENGTH =    4K       */
}

_start_of_stack = ORIGIN(RAM) + LENGTH(RAM);
```

cargo build
cargo run

Thank you .

9names, 2025-04-29:
@niekiran 
Could you please add the defmt.x linker script to your rustflags as described in the docs for defmt?
https://defmt.ferrous-systems.com/setup#linker-script