probe-rs/probe-rs #3257: Running CLI reset command results in "unreachable code" error:

jaredwolff, 2025-04-26:
### Describe the bug

When, running `probe-rs reset` without the `--chip` argument I get the following error:

```bash
thread 'tokio-runtime-worker' panicked at probe-rs/src/architecture/arm/communication_interface.rs:544:18:
internal error: entered unreachable code: Did not expect to be called with FullyQualifiedApAddress { dp: Default, ap: V2(ApV2Address(Some(2000))) }. This is a bug, please report it.
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
 WARN postcard_rpc::host_client::util: in_worker: wire receive error, exiting

thread 'main' panicked at probe-rs-tools/src/bin/probe-rs/main.rs:505:22:
called `Result::unwrap()` on an `Err` value: JoinError::Panic(Id(25), "internal error: entered unreachable code: Did not expect to be called with FullyQualifiedApAddress { dp: Default, ap: V2(ApV2Address(Some(2000))) }. This is a bug, please report it.", ...)
```

The error appears originate from the call in `cli.rs`

```bash
    let result = client
        .attach_probe(AttachRequest {
            chip: probe_options.chip,
            protocol: probe_options.protocol.map(Into::into),
            probe,
            speed: probe_options.speed,
            connect_under_reset: probe_options.connect_under_reset,
            dry_run: probe_options.dry_run,
            allow_erase_all: probe_options.allow_erase_all,
            resume_target,
        })
        .await?;
```

It's hard to tell where it fires elsewhere in the code. I'm assuming it doesn't try to determine the device type unlike Session::new.

### To Reproduce

1. Connect a Pico Probe to a nRF9151
2. Run `probe-rs` (latest 0.27 release as of today) 
3. Exhibit the error

### Expected behavior

The device should reset without failure.

❯ cargo run -- reset --chip nRF9151_xxAA
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.17s
     Running `/home/jaredwolff/Git/probe-rs/target/debug/probe-rs reset --chip nRF9151_xxAA`
Has session

### Stacktrace

```shell
❯ cargo run -- reset
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.13s
     Running `/home/jaredwolff/Git/probe-rs/target/debug/probe-rs reset`

thread 'tokio-runtime-worker' panicked at probe-rs/src/architecture/arm/communication_interface.rs:544:18:
internal error: entered unreachable code: Did not expect to be called with FullyQualifiedApAddress { dp: Default, ap: V2(ApV2Address(Some(2000))) }. This is a bug, please report it.
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>::select_ap_and_ap_bank
   3: <probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized> as probe_rs::architecture::arm::traits::DapAccess>::read_raw_ap_register
   4: probe_rs::architecture::arm::ap::memory_ap::MemoryAp::new
             at /home/jaredwolff/Git/probe-rs/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs:203:32
   5: probe_rs::architecture::arm::memory::adi_memory_interface::ADIMemoryInterface<APA>::new
             at /home/jaredwolff/Git/probe-rs/probe-rs/src/architecture/arm/memory/adi_memory_interface.rs:41:25
   6: probe_rs::architecture::arm::ap::v2::new_memory_interface
             at /home/jaredwolff/Git/probe-rs/probe-rs/src/architecture/arm/ap/v2/mod.rs:100:21
   7: <probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized> as probe_rs::architecture::arm::communication_interface::ArmProbeInterface>::memory_interface
             at /home/jaredwolff/Git/probe-rs/probe-rs/src/architecture/arm/communication_interface.rs:301:33
[... 117 lines cut ...]
             at /home/jaredwolff/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.44.2/src/runtime/scheduler/multi_thread/mod.rs:87:13
  12: tokio::runtime::context::runtime::enter_runtime
             at /home/jaredwolff/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.44.2/src/runtime/context/runtime.rs:65:16
  13: tokio::runtime::scheduler::multi_thread::MultiThread::block_on
             at /home/jaredwolff/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.44.2/src/runtime/scheduler/multi_thread/mod.rs:86:9
  14: tokio::runtime::runtime::Runtime::block_on_inner
             at /home/jaredwolff/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.44.2/src/runtime/runtime.rs:370:45
  15: tokio::runtime::runtime::Runtime::block_on
             at /home/jaredwolff/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.44.2/src/runtime/runtime.rs:340:13
  16: probe_rs::main
             at ./src/bin/probe-rs/main.rs:507:5
  17: core::ops::function::FnOnce::call_once
             at /home/jaredwolff/.rustup/toolchains/1.86.0-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

### Operating System

Linux (Ubuntu)

### Additional context

I am also using an updated nRF91xx yaml definition. Even before that, I ran into the same error (using the current one that is in the repo)

[attach_error.log](https://github.com/user-attachments/files/19918290/attach_error.log)

bugadani, 2025-04-26:
The CLI is architected as a client-server application. What you see in `cli.rs` just assembles a message, sends it to the server, and then processes the response. The top half of the stack trace points you at exactly where the panic happens.

holly-hacker, 2025-10-04:
Hi @Yatekii, I just decided to take a look myself so I happened to have the code open. I wasn't able to compile your branch (is it from an older commit on master? it tried to pull rust 1.85.0), but I put a some `dbg!` calls in the switch statement like so:
```diff
diff --git a/probe-rs/src/architecture/arm/communication_interface.rs b/probe-rs/src/architecture/arm/communication_interface.rs
index 981ad0bb..275f7ea4 100644
--- a/probe-rs/src/architecture/arm/communication_interface.rs
+++ b/probe-rs/src/architecture/arm/communication_interface.rs
@@ -457,7 +457,7 @@ impl ArmCommunicationInterface {
         let dp_state = self.select_dp(ap.dp())?;
 
         let previous_select = dp_state.current_select;
-        match (ap.ap(), &mut dp_state.current_select) {
+        match (dbg!(ap.ap()), dbg!(&mut dp_state.current_select)) {
             (ApAddress::V1(port), SelectCache::DPv1(s)) => {
                 let ap_register_address = (ap_register_address & 0xFF) as u8;
                 let ap_bank = ap_register_address >> 4;

```

This gave me the following debug output:
```rs
[probe-rs/src/architecture/arm/communication_interface.rs:460:16] ap.ap() = V2(
    ApV2Address(
        Some(
            8192,
        ),
    ),
)
[probe-rs/src/architecture/arm/communication_interface.rs:460:31] &mut dp_state.current_select = DPv1(
    SelectV1 {
        .0: 0,
        ap_sel: 0,
        ap_bank_sel: 0,
        dp_bank_sel: 0,
    },
)

```

It seems like it's using ApAddress::V2 and SelectCache::DPv1, which is a combination that isn't covered in the match statement. I have no background in this debug protocol so that's about where my useful input ends :(

jaredwolff, 2025-10-04:
@Yatekii here's the back trace from your branch

```
~/Git/probe-rs nh/debug-3257 ≡
❯ RUST_BACKTRACE=1 cargo run --bin probe-rs -- reset
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.14s
     Running `target/debug/probe-rs reset`

thread 'tokio-runtime-worker' panicked at probe-rs/src/architecture/arm/communication_interface.rs:472:18:
internal error: entered unreachable code: Did not expect to be called with V2(ApV2Address(Some(2000))) and DPv1(SelectV1 { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }). This is a bug, please report it.
stack backtrace:
   0: __rustc::rust_begin_unwind
             at /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/std/src/panicking.rs:697:5
   1: core::panicking::panic_fmt
             at /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/core/src/panicking.rs:75:14
   2: select_ap_and_ap_bank
             at ./probe-rs/src/architecture/arm/communication_interface.rs:472:18
   3: <probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface as probe_rs::architecture::arm::traits::DapAccess>::read_raw_ap_register
             at ./probe-rs/src/architecture/arm/communication_interface.rs:547:9
   4: new<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface>
             at ./probe-rs/src/architecture/arm/ap/memory_ap/mod.rs:206:32
   5: probe_rs::architecture::arm::memory::adi_memory_interface::ADIMemoryInterface<APA>::new
             at ./probe-rs/src/architecture/arm/memory/adi_memory_interface.rs:39:25
   6: new_memory_interface
             at ./probe-rs/src/architecture/arm/ap/v2/mod.rs:99:21
   7: memory_interface
             at ./probe-rs/src/architecture/arm/communication_interface.rs:265:33
   8: probe_rs::architecture::arm::sequences::ArmDebugSequence::debug_core_start
             at ./probe-rs/src/architecture/arm/sequences.rs:730:24
   9: probe_rs::core::core_state::CombinedCoreState::enable_arm_debug::{{closure}}
             at ./probe-rs/src/core/core_state.rs:124:13
  10: in_scope<probe_rs::core::core_state::{impl#0}::enable_arm_debug::{closure_env#0}, core::result::Result<(), probe_rs::architecture::arm::ArmError>>
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tracing-0.1.41/src/span.rs:1102:9
  11: enable_arm_debug
             at ./probe-rs/src/core/core_state.rs:122:9
  12: attach_arm_debug_interface
             at ./probe-rs/src/session.rs:322:17
  13: probe_rs::session::Session::new
             at ./probe-rs/src/session.rs:177:13
  14: probe_rs::probe::Probe::attach_with_registry
             at ./probe-rs/src/probe.rs:349:9
  15: probe_rs::probe::Probe::attach
             at ./probe-rs/src/probe.rs:334:9
  16: probe_rs::util::common_options::LoadedProbeOptions::attach_session
             at ./probe-rs-tools/src/bin/probe-rs/util/common_options.rs:311:13
  17: {async_fn#0}
             at ./probe-rs-tools/src/bin/probe-rs/rpc/functions/probe.rs:242:23
  18: {async_fn#0}
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/postcard-rpc-0.11.15/src/server/dispatch_macro.rs:221:29
  19: {async_fn#0}<probe_rs::rpc::transport::memory::WireTx<tokio::sync::mpsc::bounded::Sender<alloc::vec::Vec<u8, alloc::alloc::Global>>>, probe_rs::rpc::transport::memory::WireRx<tokio::sync::mpsc::bounded::Receiver<core::result::Result<alloc::vec::Vec<u8, alloc::alloc::Global>, postcard_rpc::server::WireRxErrorKind>>>, alloc::boxed::Box<[u8], alloc::alloc::Global>, probe_rs::rpc::functions::impls::RpcApp<2>>
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/postcard-rpc-0.11.15/src/server/mod.rs:457:33
  20: {async_block#0}
             at ./probe-rs-tools/src/bin/probe-rs/main.rs:508:63
  21: poll<alloc::boxed::Box<probe_rs::main::{async_block#0}::{async_block_env#0}, alloc::alloc::Global>>
             at /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/core/src/future/future.rs:124:9
  22: tokio::runtime::task::core::Core<T,S>::poll::{{closure}}
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.47.1/src/runtime/task/core.rs:365:17
  23: with_mut<tokio::runtime::task::core::Stage<core::pin::Pin<alloc::boxed::Box<probe_rs::main::{async_block#0}::{async_block_env#0}, alloc::alloc::Global>>>, core::task::poll::Poll<postcard_rpc::server::ServerError<probe_rs::rpc::transport::memory::WireTx<tokio::sync::mpsc::bounded::Sender<alloc::vec::Vec<u8, alloc::alloc::Global>>>, probe_rs::rpc::transport::memory::WireRx<tokio::sync::mpsc::bounded::Receiver<core::result::Result<alloc::vec::Vec<u8, alloc::alloc::Global>, postcard_rpc::server::WireRxErrorKind>>>>>, tokio::runtime::task::core::{impl#6}::poll::{closure_env#0}<core::pin::Pin<alloc::boxed::Box<probe_rs::main::{async_block#0}::{async_block_env#0}, alloc::alloc::Global>>, alloc::sync::Arc<tokio::runtime::scheduler::multi_thread::handle::Handle, alloc::alloc::Global>>>
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.47.1/src/loom/std/unsafe_cell.rs:16:9
  24: poll<core::pin::Pin<alloc::boxed::Box<probe_rs::main::{async_block#0}::{async_block_env#0}, alloc::alloc::Global>>, alloc::sync::Arc<tokio::runtime::scheduler::multi_thread::handle::Handle, alloc::alloc::Global>>
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.47.1/src/runtime/task/core.rs:354:13
  25: tokio::runtime::task::harness::poll_future::{{closure}}
[... cut ...]
  15: block_on<probe_rs::main::{async_block_env#0}>
             at /home/jared/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.47.1/src/runtime/runtime.rs:328:13
  16: probe_rs::main
             at ./probe-rs-tools/src/bin/probe-rs/main.rs:517:5
  17: core::ops::function::FnOnce::call_once
             at /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

Yatekii, 2025-10-04:
Thanks a lot! This is what I expected. We do not switch the DP correctly.

@xobs @ryan-summers: you were the ones working the most with this. Any clues where we might not switch properly?
I see that we do not switch when the DP is already in the cache - I am not sure that's correct. I would expect us having to switch - but not init - if it's already in the cache?

piersfinlayson, 2025-11-09:
I am hitting a very similar error message with probe-rs 0.30.0, in my application:

```
thread 'tokio-runtime-worker' (28908) panicked at /home/pdf/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/probe-rs-0.30.0/src/architecture/arm/communication_interface.rs:472:18:
internal error: entered unreachable code: Did not expect to be called with FullyQualifiedApAddress { dp: Default, ap: V1(0) }. This is a bug, please report it.
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

In my case I am using an STM32F103CBTx based CMSIS-DAP probe running DAPLink.  My target is an RP2350.

In this code explicitly, I have told probe-rs to attempt to read an STM32F411RETx chip (and it's an RP2350).  I would expect a failure, but not a panic.

In case you're wondering why I'm attempting to read as an STM32F4 - my target hardware comes in STM32F4 and RP2350 variants, and I am trying to autodetect, using two different types of reads from probe-rs.

Happy to collect more diagnostics - let me know - I will experiment to see if I can make this use case work without panicing.

piersfinlayson, 2025-11-09:
I have worked around this by replacing the panic with

            _ => return Err(
                ArmError::Other(
                    format!("Unexpected AP / DP combination {}/{}", ap.ap(), dp_state.debug_port_version)
                )
            ),

Arm(Other("Unexpected AP / DP combination V1(0)/DPv3"))

This is the sort of error I'd expect in this scenario - I've told probe-rs it's a V1 AP (STM32F4) yet, presumably, probe-rs has detected something DPv3y from the RP2350.

I haven't checked the unwinding is fully graceful, but it now seems to work for my scenario.

It's also possible that probe-rs should have explicitly detected the mismatch and errored out earlier - I'm not familiar enough with the code to know.