probe-rs/probe-rs #3305: symbols and breakpoint error

mortifia, 2025-05-11:
### Describe the bug

[esp32c6] [Windows 11] [VSCode] [Rust nightly]

Summary:
Breakpoints work initially, but source mapping does not. After enabling remap-cwd-prefix, breakpoints stop working entirely, although debug symbols are still visible when paused. This issue appears to be specific to Windows.

The issue seems to come from the path for the symbols

[image]

`readelf -wL .\target\riscv32imac-unknown-none-elf\debug\vr_tracker`
```
...
src\bin/main.rs:
main.rs                                   41          0x420054d4      38       x
main.rs                                   42          0x420054ec      39       x
main.rs                                   45          0x42005504      40       x
main.rs                                   49          0x42005528      41       x
main.rs                                   51          0x42005534      42       x
main.rs                                   57          0x42005542      43       x
main.rs                                   29          0x42005544      44       x
main.rs                                   61          0x42005554      45       x
main.rs                                   58          0x42005556      46       x
main.rs                                   61          0x42005580      47       x
main.rs                                   61          0x420055f2      48
main.rs                                    0          0x420055f4      49
main.rs                                   61          0x420055f8      50       x
main.rs                                   61          0x42005614      51
main.rs                                   62          0x4200561c      52       x
main.rs                                   65          0x42005620      53       x
main.rs                                   65          0x42005638      54
main.rs                                   65          0x42005678      55
main.rs                                   69          0x4200568e      56       x
main.rs                                   69          0x4200569e      57
main.rs                                   69          0x420056ba      58
main.rs                                   69          0x420056ca      59
...

src\mem/maybe_uninit.rs:
maybe_uninit.rs                          308          0x42005890       9       x
maybe_uninit.rs                          475          0x420058a4      10       x

src/lib.rs:
lib.rs                                   143          0x420058bc      11       x
lib.rs                                     -          0x420058c6

lib.rs                                    89          0x420058c6               x
lib.rs                                    90          0x420058d6       1       x
lib.rs                                    92          0x420058f4       2       x

src/ub_checks.rs:
ub_checks.rs                              74          0x42005902       3       x
ub_checks.rs                               0          0x4200590e       4
ub_checks.rs                              75          0x42005916       5       x
ub_checks.rs                               0          0x42005920       6

src\ptr/mut_ptr.rs:
mut_ptr.rs                              1056          0x42005926       7       x
...
```

`launch.json`
```
{
    "version": "0.2.0",
    "configurations": [
       {
        "type": "probe-rs-debug",
        "request": "launch",
        "name": "probe-rs Test",
        "cwd": "${workspaceFolder}",
        "connectUnderReset": false,
        "chip": "esp32c6",
        "flashingConfig": {
            "flashingEnabled": true,
            "haltAfterReset": true
        },
        "probe": "303a:1001:F0:F5:BD:27:56:CC",
        "coreConfigs": [
            {
                "svdFile": "${workspaceFolder}/esp32c6.svd",
                "coreIndex": 0,
                // "programBinary": "${workspaceFolder}/target/riscv32imac-unknown-none-elf/debug/${workspaceFolderBasename}",
                "programBinary": "${workspaceFolder}/target/riscv32imac-unknown-none-elf/debug/vr_tracker"

            }
        ]
       }
    ]
}
```

after changes in `.cargo\config.toml` 
```
[build]
rustflags = [
  "-C", "force-frame-pointers",
  "-Z", "remap-cwd-prefix=."  # added
  ]
```

breakpoint stop worked 
```
Cannot set breakpoint here. Try reducing compile time-, and link time-, optimization in your build configuration, or choose a different source location: No valid breakpoint information found for file: d:\dev\vr_tracker\src\bin\main.rs, line: 65, column: Some(1)
```
[image]

but when i pause i have the symbols 

[image]

### To Reproduce

1. Clone `https://github.com/mortifia/vr_tracker`
2. possibly needed to build
> powershell
```powershell
    irm https://github.com/probe-rs/probe-rs/releases/latest/download/probe-rs-tools-installer.ps1 | iex
```
2. build in nightly
3. add a breakpoint in main.rs
3. use the launch.json 

### Expected behavior

Have working symbols on breakpoint

### Stacktrace

```shell

```

### Operating System

Windows

bugadani, 2025-05-11:
The source mapping issue was likely fixed by https://github.com/probe-rs/probe-rs/pull/3137 - the breakpoints are likely related, but they have other problems, some of that are being addressed currently.

mortifia, 2025-05-11:
The future version work with this weird path ?   `src\mem/maybe_uninit.rs`

Yatekii, 2025-05-19:
That path you mention should already work - provided we actually get the symbol info for it. If you have an issue with it despite symbols being present, that'd be another issue I guess.