probe-rs/probe-rs #3316: probe-rs run: RP2040 Failed to switch to DP Multidrop on Ctrl+C

rursprung, 2025-05-16:
### Describe the bug

every time i hit Ctrl+C when running a program on an RP2040 i get the following warning:
```
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(11002927): An error occurred in the communication with an access port or debug port.
```

tested with both probe-rs 0.27.0 (git commit: 7d59a31) as well as the current git version (git commit: 5fc304e0).

output from the current git version:
```
$ cargo run -- run --chip RP2040 path/to/firmware
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 1.16s
     Running `target\debug\probe-rs.exe run --chip RP2040 path/to/firmware`
      Erasing ✔ 100% [####################] 176.00 KiB @  94.54 KiB/s (took 2s)
  Programming ✔ 100% [####################] 176.00 KiB @  41.46 KiB/s (took 4s)
     Finished in 6.19s
[INFO ] setup done (my_firmware src/bin/main.rs:42)
Received Ctrl+C, exiting
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(11002927): An error occurred in the communication with an access port or debug port.
Error: Exited due to user request
error: process didn't exit successfully: `target\debug\probe-rs.exe run --chip RP2040 path/to/firmware` (exit code: 1)
```

### To Reproduce

1. connect a picoprobe to an RP Pico
2. use `probe-rs run` (in my case as a runner for `cargo run`) to run some application on an RP2040 (in my case an RP Pico)
3. press Ctrl+C after it has launched

### Expected behavior

no warnings

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2025-05-16:
is this a problem because of the warning, or is this a problem because of the non-zero return code?

rursprung, 2025-05-16:
> is this a problem because of the warning, or is this a problem because of the non-zero return code?

i found it confusing since pressing Ctrl+C is the obvious (only?) way to get out of it, so i'd expect a clean exit with neither a warning nor a non-zero exit code.
i'm happy to ignore it myself for now, though if i'd not use an RP2040 in a while and then get back to it i might've forgotten about this peculiarity and then wonder again why i get the warning. also i asked in the chat yesterday and it was suggested that i should report it for tracking purposes 😉

bugadani, 2025-05-16:
definitely, thanks for that. will think of something

grtwje, 2025-09-21:
I'll add a "me too" to this. I was dusting off an old project and updating everything to the latest and greatest and encountered this error message. I was wondering if I had messed up the update until I saw this issue.