probe-rs/probe-rs #3326: Stabilize RPC API

bugadani, 2025-05-17:
### Is your feature request related to a problem? Please describe.

For the server/client split to be useful, we need to figure out how to best organize messages or maybe add functionality to postcard-rpc, so that we don't need to break compatibility often. Currently it's possible to tie multiple topics together and treat them like a single enum that can be expanded with new variants, but this needs extra work to prevent reading messages out of order.

Yatekii, 2025-05-17:
If we are at this, I wanna throw one concern into the ring: postcard-rpc is not really a thing in other languages. Also, I am not sure how easy it is to add fields to specific requests after the fact. So I think it's worth it to throw a protocol that's a bit more universal into the ring imo. I must say, postcard-rpc felt very nice and I don't want to get rid of it.

bugadani, 2025-05-17:
I'm hoping @jamesmunns can chime in with opinions/advice but I'm personally not worried about lack of language support. Future additions are a concern of mine as well.

jamesmunns, 2025-05-18:
> I am not sure how easy it is to add fields to specific requests after the fact

Generally, you don't. "all changes are breaking changes", BUT that's only on a per-API basis (e.g. changing one endpoint doesn't "break" other endpoints). My suggested usage is "don't", and instead add new endpoints. A server (or client) could choose to support multiple versions of the same API, and use service discovery to detect which one to use at runtime.

Basically: you can't CHANGE an endpoint, but it is very easy to simultaneously support v1, v2, v3, and v4 of an endpoint.

> postcard-rpc is not really a thing in other languages

Also very true. I have some initial work on `postcard-idl` which aims to allow codegen for other langs, but it's definitely a "at some point in the future" thing.

Yatekii, 2025-05-18:
> Basically: you can't CHANGE an endpoint, but it is very easy to simultaneously support v1, v2, v3, and v4 of an endpoint.

That's probably the desired scenario anyways :)

> but it's definitely a "at some point in the future" thing

Good enough for me tbh :)

bugadani, 2025-05-18:
> Generally, you don't.

I wonder if this is a case of "trick users into correctness". On one hand, I'm worried that we'll have weird endpoint versions that really only differ in additive enum variants, and code that facilitates this (e.g. by simply not emitting an event for an "unsupported in this version" enum variant, and adding a conversion, so that implementations don't need to be duplicated. On the other hand, this is a great way to not introduce a breaking change - because no changes can be introduced. I guess we'll see how this looks once we consider the API stable.

jamesmunns, 2025-06-02:
Scanning back through my emails, and wanted ping @bugadani about:

> trick users (that is, us) into correctness

Just wanted to see if you could expand on that a bit: the stated goal is "misuse resistance", e.g. "you can't hold it wrong accidentally", but I wanted to make sure what you meant about tricking users. It's true there's no "I know what I'm doing" escape hatch, but that's a little bit by design :)

(also to be clear: no offense taken! Just wanted to make sure I understood the feedback)

closed 2026-08-28 by bugadani