probe-rs/probe-rs #3333: Implement a DAP plugin for Zed

bugadani, 2025-05-17:


Vollbrecht, 2025-06-04:
for reference we talk about interacting with [this](https://github.com/zed-industries/zed/blob/main/crates/project/src/debugger.rs) implementation ?

bugadani, 2025-06-04:
I don't know, haven't looked into this yet. This issue was created solely based on "zed supports the DAP debug protocol" headline, nothing more.

casaroli, 2025-06-09:
I took a quick look (a failed shot) at this and I think we have at least some problems:

- `probe-rs` binary from `probe-rs-tools` dap_server only supports TCP and it seems zed is expecting the process to use stdio. This is not difficult to fix in `probe-rs` side.
- some parameters from probe-rs (especially `coreConfigs`) would need to be passed down to probe-rs DAP
- Zed tried to send the command `setExceptionBreakpoints` which is not supported by `probe-rs` (yet?)
- what else do we need to send in DAP that zed is not sending?

Maybe we need to create a probe-rs dap adapter and have the launch json schema integrated there.

What do you think?

bugadani, 2025-06-21:
Guess we can close this now, see https://github.com/probe-rs/probe-rs/pull/3419 and https://github.com/Tiwalun/probe-rs-zed

closed 2025-06-21 by bugadani