probe-rs/probe-rs #3337: No defmt_rtt output in VSCode with 0.28.0 (rppico + picoprobe)

jenshnielsen, 2025-05-18:
### Describe the bug

When I run a simple example such as https://github.com/rp-rs/rp2040-project-template with 0.28.0 I observe no defmt_rtt output which works with 0.27.0

I have bisected the issue to https://github.com/probe-rs/probe-rs/pull/3182 and changing the default value of `disallow_clearing_rtt_header` back to true makes 0.28.0 work as expected again. 

I looked for a way to change the value of `disallow_clearing_rtt_header` but I cannot see it being available in the config file. Perhaps I overlooked something?

defmt_rtt output works with both 0.27.0 and 0.28.0 when using cargo run from the command line and is not affected by this.

### To Reproduce

* clone https://github.com/rp-rs/rp2040-project-template
* Conneted to a picoprobe and rp2040 
* Open vscode and run in debug mode (f5) 
* Observe that the code runs and debugging works but no defmt_rtt output is printed in the vscode terminal
* Downgrade to 0.27.0 and observe that output works again

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2025-05-18:
I can try cloning the repo but before I try anything: the template doesn't specify any RTT channels and what format they'd use. Are you sure I don't need to modify anything in the template?

jenshnielsen, 2025-05-18:
@bugadani Yes the example works as is with 0.27.0 and produces the defmt_rtt output

I tried adding a channel config like this to lauch.json with 0.28.0 but it does not seem to make a difference.

```
                    "rttChannelFormats": [
                        {
                            "channelNumber": 0,
                            // Format RTT data as String data
                            "dataFormat": "String",
                            // Include host-side timestamps for every line of data transferred from the target RTT output
                            "showTimestamps": true
                        },
                    ]
```

With 0.28.0 I see the following output from probe-rs

```
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 51337
probe-rs-debug: Starting debug session from: 127.0.0.1:51345
FLASHING: Starting write of "C:\\Users\\Jens\\source\\repos\\kitronik\\rp2040-project-template\\target/thumbv6m-none-eabi/debug/rp2040-project-template" to device memory

Triggered DAP Event: progressStart
10

Triggered DAP Event: progressUpdate

Triggered DAP Event: progressEnd
FLASHING: Completed write of "C:\\Users\\Jens\\source\\repos\\kitronik\\rp2040-project-template\\target/thumbv6m-none-eabi/debug/rp2040-project-template" to device memory
   Sent DAP Response sequence #2 : launch

Triggered DAP Event: initialized

Received DAP Request sequence #3 : setBreakpoints
   Sent DAP Response sequence #3 : setBreakpoints

Received DAP Request sequence #4 : setInstructionBreakpoints
   Sent DAP Response sequence #4 : setInstructionBreakpoints

Received DAP Request sequence #5 : configurationDone
   Sent DAP Response sequence #5 : configurationDone

Received DAP Request sequence #6 : threads
   Sent DAP Response sequence #6 : threads
```

But with 0.27.0 I see the following:

```
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 52031
probe-rs-debug: Starting debug session from: 127.0.0.1:52035
FLASHING: Starting write of "C:\\Users\\Jens\\source\\repos\\kitronik\\rp2040-project-template\\target/thumbv6m-none-eabi/debug/rp2040-project-template" to device memory

Triggered DAP Event: progressStart
10

Triggered DAP Event: progressUpdate

Triggered DAP Event: progressEnd
FLASHING: Completed write of "C:\\Users\\Jens\\source\\repos\\kitronik\\rp2040-project-template\\target/thumbv6m-none-eabi/debug/rp2040-project-template" to device memory
   Sent DAP Response sequence #2 : launch

Triggered DAP Event: initialized

Received DAP Request sequence #3 : setBreakpoints
   Sent DAP Response sequence #3 : setBreakpoints

Received DAP Request sequence #4 : setInstructionBreakpoints
   Sent DAP Response sequence #4 : setInstructionBreakpoints

Received DAP Request sequence #5 : configurationDone
   Sent DAP Response sequence #5 : configurationDone

Triggered DAP Event: probe-rs-rtt-channel-config
probe-rs-debug: Opened a new RTT Terminal window named: defmt

Received DAP Request sequence #6 : threads
   Sent DAP Response sequence #6 : threads

Received DAP Request sequence #7 : rttWindowOpened
probe-rs-debug: RTT Window opened, and ready to receive RTT data on channel 0
   Sent DAP Response sequence #7 : rttWindowOpened
62

Triggered DAP Event: probe-rs-rtt-data

```

E.g. it looks like 

```
Triggered DAP Event: probe-rs-rtt-channel-config
```
Is triggered with 0.27.0 but not with 0.28.0

bugadani, 2025-05-19:
Okay I can reproduce this issue and it is caused by the debugger clearing the control block too late. As a workaround, you can enable `haltAfterReset: true` until this gets resolved.

closed 2025-05-19 by bugadani

jenshnielsen, 2025-05-20:
Thanks @bugadani I can confirm that #3351 resolves the issue for me and that `haltAfterReset` also works as a workaround for `0.28.0`

PR #3351 (2025-05-19): RTT fixes
Fixes #3342
Fixes #3337