probe-rs/probe-rs #3342: nrf5340: no RTT output on network core

0e4ef622, 2025-05-18:
### Describe the bug

Starting from v0.28.0, probe-rs is unable to read RTT logs when attaching to the network core. I believe this is due to probe-rs attempting to read through the wrong core. Previous versions had code to deduce which core to use based on the memory address, but it seems this was [removed](https://github.com/probe-rs/probe-rs/pull/3049/files#diff-eab702ba8f2e5704ef303937829288e4cbcc50b5ea482f38db04c5685b2f3ffaL56) and forgotten about.

### To Reproduce

1. Clone https://github.com/0e4ef622/embassy-template (needs nRF Command Line Tools)
2. `cd nrf53`
3. `just flash-recover`
4. `just attach-net`

### Expected behavior

A hello world message appears.

### Stacktrace

```shell

```

### Operating System

Other Linux Distribution

### Additional context

Here's a snippet of the repeating portion of the logs from running `RUST_LOG="probe_rs=trace" probe-rs attach --chip nRF5340_xxAA build/net-core.elf`

```
 INFO probe_rs::architecture::arm::traits::polyfill: Error in access 1/8 of block access: FaultResponse
DEBUG probe_rs::architecture::arm::traits::polyfill: Clearing overrun and sticky error
TRACE probe_rs::probe::jlink: Buffer length for j-link transfer: 18
TRACE probe_rs::probe::jlink: write 18 bytes: [cf, 0, 38, 0, ff, e0, ff, ff, ff, ff, ff, 81, 80, 2, 0, 0, 0, 0]
TRACE probe_rs::probe::jlink: read 8 bytes: [0, 19, 0, 0, 0, 0, 0, 0]
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 0: Ok 0
DEBUG status: probe_rs::architecture::arm::memory::adi_memory_interface: Reading chunk with len 1 at address 0xe000edf0
DEBUG status: probe_rs::architecture::arm::ap: Writing AP register TAR, value=TAR { address: e000edf0 }
DEBUG status: probe_rs::architecture::arm::traits::polyfill: Performing 2 transfers (1 additional transfers)
TRACE status: probe_rs::probe::jlink: Buffer length for j-link transfer: 32
TRACE status: probe_rs::probe::jlink: write 32 bytes: [cf, 0, 6e, 0, ff, e0, ff, ff, ff, ff, ff, ff, 0, 0, 0, 0, c0, 3f, 8b, 0, be, 1d, 0, 3c, 0, bd, 0, 0, 0, 0, 0, 0]
TRACE status: probe_rs::probe::jlink: read 15 bytes: [0, 19, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 30, 0, 0]
TRACE status: probe_rs::architecture::arm::traits::polyfill: Transfer result 0: Ok 0
TRACE status: probe_rs::architecture::arm::traits::polyfill: DAP read 0.
TRACE status: probe_rs::architecture::arm::traits::polyfill: Transfer result 1: Ok 0
DEBUG status:read_ap_register: probe_rs::architecture::arm::traits::polyfill: Performing 2 transfers (1 additional transfers) ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::probe::jlink: Buffer length for j-link transfer: 30 ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::probe::jlink: write 30 bytes: [cf, 0, 64, 0, ff, 0, 0, 0, 0, c0, 3f, 0, 0, 0, 0, f0, f, 9f, 0, 0, 0, 0, 40, 2f, 0, 0, 0, 0, 0, 0] ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::probe::jlink: read 14 bytes: [0, 1, 0, 0, 0, 30, 40, 2, 0, 28, 0, e, 0, 0] ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::architecture::arm::traits::polyfill: DAP read 0. ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::architecture::arm::traits::polyfill: Transfer result 0: Ok 0 ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::architecture::arm::traits::polyfill: DAP read 1310721. ap=0 register="DRW"
TRACE status:read_ap_register: probe_rs::architecture::arm::traits::polyfill: Transfer result 1: Ok 140001 ap=0 register="DRW"
DEBUG status:read_ap_register: probe_rs::architecture::arm::ap: Register read succesful ap=0 register="DRW" value=1310721
DEBUG status: probe_rs::architecture::arm::memory::adi_memory_interface: Finished reading block
DEBUG probe_rs::architecture::arm::memory::adi_memory_interface: Reading chunk with len 8 at address 0x21000020
DEBUG probe_rs::architecture::arm::ap: Writing AP register TAR, value=TAR { address: 21000020 }
DEBUG probe_rs::architecture::arm::traits::polyfill: Performing 2 transfers (1 additional transfers)
TRACE probe_rs::probe::jlink: Buffer length for j-link transfer: 32
TRACE probe_rs::probe::jlink: write 32 bytes: [cf, 0, 6e, 0, ff, e0, ff, ff, ff, ff, ff, ff, 0, 0, 0, 0, c0, 3f, 8b, 0, 4, 0, 20, 24, 0, bd, 0, 0, 0, 0, 0, 0]
TRACE probe_rs::probe::jlink: read 15 bytes: [0, 19, 0, 0, 0, 0, 0, 0, 9, 0, a0, 0, 38, 0, 0]
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 0: Ok 0
TRACE probe_rs::architecture::arm::traits::polyfill: DAP read 1310721.
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 1: Ok 140001
DEBUG probe_rs::architecture::arm::ap: Reading register DRW, block with len=8 words
DEBUG probe_rs::architecture::arm::traits::polyfill: Performing 9 transfers (1 additional transfers)
TRACE probe_rs::probe::jlink: Buffer length for j-link transfer: 110
TRACE probe_rs::probe::jlink: write 110 bytes: [cf, 0, a6, 1, ff, 0, 0, 0, 0, c0, 3f, 0, 0, 0, 0, f0, f, 0, 0, 0, 0, fc, 3, 0, 0, 0, 0, ff, 0, 0, 0, 0, c0, 3f, 0, 0, 0, 0, f0, f, 0, 0, 0, 0, fc, 3, 0, 0, 0, 0, ff, 0, 0, 0, 0, c0, 3f, 9f, 0, 0, 0, 0, c0, 27, 0, 0, 0, 0, f0, 9, 0, 0, 0, 0, 7c, 2, 0, 0, 0, 0, 9f, 0, 0, 0, 0, c0, 27, 0, 0, 0, 0, f0, 9, 0, 0, 0, 0, 7c, 2, 0, 0, 0, 0, bd, 0, 0, 0, 0, 0, 0]
TRACE probe_rs::probe::jlink: read 54 bytes: [0, 9, 0, a0, 0, 38, 0, ff, ff, ff, ff, f, c0, ff, ff, ff, ff, 3, f0, ff, ff, ff, ff, 0, fc, ff, ff, ff, 3f, 0, ff, ff, ff, ff, f, c0, ff, ff, ff, ff, 3, f0, ff, ff, ff, ff, 0, fc, ff, ff, ff, 3f, 0, 0]
TRACE probe_rs::architecture::arm::traits::polyfill: DAP read 1310721.
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 0: Ok 140001
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 1: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 2: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 3: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 4: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 5: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 6: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 7: Failed(FaultResponse) 0
TRACE probe_rs::architecture::arm::traits::polyfill: Transfer result 8: Failed(FaultResponse) 0
DEBUG probe_rs::architecture::arm::traits::polyfill: Transfer 2/9 failed: Failed(FaultResponse)
 INFO probe_rs::architecture::arm::traits::polyfill: Error in access 1/8 of block access: FaultResponse
...
```

bugadani, 2025-05-19:
Apologies. Can you please test https://github.com/probe-rs/probe-rs/pull/3351 for me?

0e4ef622, 2025-05-19:
Can confirm that PR fixes it!

PR #3351 (2025-05-19): RTT fixes
Fixes #3342
Fixes #3337
closed 2025-05-19 by bugadani