probe-rs/probe-rs #3378: Indicated that `probe-rs info` failures are not indicative of something being wrong

Yatekii, 2025-05-26:
### Is your feature request related to a problem? Please describe.

We repreatedly have people coming up asking why their chip is not working when it's just `probe-rs info` not caughing up sane data.

### Describe the solution you'd like

Add big red letters that explain that everything might be working even if `probe-rs info` does not.

bugadani, 2025-05-26:
Let's just hide probe-rs info behind a debug flag...

Yatekii, 2025-05-26:
I feel like it can actually be useful to find out what stuff your chip has, but it's currently more of a debug feature than sth reliable, I agree. I see us being flooded with "info does not work anymore" type of issues/requests though :D