probe-rs/probe-rs #3408: Unable to debug HardFault on STM32U5

dimpolo, 2025-06-16:
### Describe the bug

When encountering a HardFault probe-rs is unable to produce a stack trace.
The following is printed:
```text
Firmware exited unexpectedly: Exception
 WARN probe_rs_debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Probe(Register("No Stack Pointer register. Please report this as a bug."))
Core 0
    Frame 0: HardFault_ @ 0x080003fc
       C:\Users\d.polonski\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-rt-0.7.5\src\lib.rs:1105:5
    Frame 1: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Probe(Register("No Stack Pointer register. Please report this as a bug.")) @ 0x080003fc
```

### To Reproduce

main.rs
```rust
#![no_main]
#![no_std]

extern crate embassy_stm32;

use core::panic::PanicInfo;

#[cortex_m_rt::entry]
fn main() -> ! {
    cortex_m::asm::udf();
}

#[panic_handler]
fn panic(_: &PanicInfo) -> ! {
    loop {}
}
```

Cargo.toml
```toml
[package]
name = "test"
version = "0.0.0"
edition = "2021"

[dependencies]
cortex-m-rt = { version = "0.7.5" }
cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy.git", rev = "dda17cc", features = ["memory-x", "stm32u595ri"] }
```

.cargo/config.toml
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-rs run --chip STM32U595RITx"

rustflags = [
    "-C", "link-arg=-Tlink.x",
]

[build]
target = "thumbv8m.main-none-eabihf"
```

### Stacktrace

```shell

```

### Operating System

None

bugadani, 2025-06-16:
Although your output is a bit different, I think this is a dup of https://github.com/probe-rs/probe-rs/issues/3309

dimpolo, 2025-06-16:
> Someone needs to teach probe-rs to unwind beyond exception frames.

It can, see #1665

I found the issue:
The stack pointer id need to be `13` not `0b00011000`
https://github.com/probe-rs/probe-rs/blob/d8915ee4d05c8998e75c3a3d6732b6b852856c50/probe-rs-debug/src/exception_handling/armv8m.rs#L408

https://github.com/probe-rs/probe-rs/blob/d8915ee4d05c8998e75c3a3d6732b6b852856c50/probe-rs/src/architecture/arm/core/registers/cortex_m.rs#L26-L32

@noppej @sullivanmj could you have a look?

sullivanmj, 2025-07-29:
I meant to respond to this a while back, but it fell off my radar. I was able to reproduce the problem, but my first-stab attempt at the fix under consideration did not work for me. I don't know if I did something wrong or not, and unfortunately won't likely have time to investigate it any further in the near future.

natevw, 2025-07-30:
Bummer but thanks for the update @sullivanmj! I just happened to start hitting this today (on a U575 chip) and found this issue. Was going to mention that I was also seeing quite a few occurrences of:

> WARN probe_rs::probe::stlink:733: send_jtag_command 242 failed: SwdApWdataError

but based on the discussion above maybe that's an unrelated thing. Sounds like any `armv8m` target will hit this issue until support for that platform gets implemented more fully/correctly?

closed 2025-12-23 by bugadani