probe-rs/probe-rs #3414: Error communicating with the probe.

Caused by:
    0: An ARM specific error occurred.
    1: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    2: Failed to read register DRW at address 0xd0c
    3: An error occurred in the co

kaspar030, 2025-06-19:
```json
{
  "meta": {
    "version": "0.27.0",
    "commit": "0.27.0 (git commit: v0.27.0-18-g9ef35402)",
    "arch": "x86_64",
    "os": "linux"
  },
  "command": [
    "target/debug/probe-rs",
    "run",
    "--protocol=swd",
    "--chip",
    "nrf5340_xxAA",
    "--preverify",
    "--allow-erase-all",
    "/home/kaspar/src/own/rust/ariel-os/build/bin/nrf5340dk-net/cargo/thumbv8m.main-none-eabi/release/hello-world",
    "-r=report.json"
  ],
  "elf": "/home/kaspar/src/own/rust/ariel-os/build/bin/nrf5340dk-net/cargo/thumbv8m.main-none-eabi/release/hello-world",
  "log": "/home/kaspar/.local/share/probe-rs/1750334527686.log",
  "error": "Error communicating with the probe.\n\nCaused by:\n    0: An ARM specific error occurred.\n    1: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.\n    2: Failed to read register DRW at address 0xd0c\n    3: An error occurred in the communication with an access port or debug port.\n    4: Target device responded with a FAULT response to the request."
}
```

Here's the trace with `RUST_LOG=probe_rs=trace`: (warning, expands to ~120MB!)

[1750334527686.log.gz](https://github.com/user-attachments/files/20818726/1750334527686.log.gz)

Tiwalun, 2025-07-24:
Could you please try the latest master branch? I think #3491 might have fixed this.

Tiwalun, 2025-07-26:
@ROMemories , that doesn't seem to be the same error. The original trace failed because it tries to access RAM with the wrong core, it tries to access RAM which belongs to the network core by using the application core, and that fails.

In your case it seems to flash fine, and the firmware seems to run, as indicated by `Firmware exited unexpectedly: Exception.
What are the steps to reproduce the problem you're seeing? Best would be a firmware which runs on the DK.
`