probe-rs/probe-rs #3455: probe-rs link ch32v003 failed

xiaguangbo, 2025-07-08:
### Describe the bug

.cargo/config.toml:
```
[build]
target = "riscv32ec-unknown-none-elf.json"

[target."riscv32ec-unknown-none-elf"]
runner = "probe-rs run --chip ch32v003"

[unstable]
build-std = ["core"]
```

```
xiaguangbo@fedora:~/xiaguangbo/linux/project/riscx32/program_riscx32/project$ cargo run --release
    Finished `release` profile [optimized] target(s) in 0.08s
     Running `probe-rs run --chip ch32v003 target/riscv32ec-unknown-none-elf/release/project`

thread 'tokio-runtime-worker' panicked at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:526:2:
cannot load 32 bits from a 64-bit region
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
 WARN postcard_rpc::host_client::util: in_worker: wire receive error, exiting

thread 'main' panicked at probe-rs-tools/src/bin/probe-rs/main.rs:517:22:
called `Result::unwrap()` on an `Err` value: JoinError::Panic(Id(25), "cannot load 32 bits from a 64-bit region", ...)
```

### Stacktrace

```shell
xiaguangbo@fedora:~/xiaguangbo/linux/project/riscx32/program_riscx32/project$ cargo run --release
    Finished `release` profile [optimized] target(s) in 0.15s
     Running `probe-rs run --chip ch32v003 target/riscv32ec-unknown-none-elf/release/project`

thread 'tokio-runtime-worker' panicked at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:526:2:
cannot load 32 bits from a 64-bit region
stack backtrace:
   0: __rustc::rust_begin_unwind
             at /rustc/17067e9ac6d7ecb70e50f92c1944e545188d2359/library/std/src/panicking.rs:697:5
   1: core::panicking::panic_fmt
             at /rustc/17067e9ac6d7ecb70e50f92c1944e545188d2359/library/core/src/panicking.rs:75:14
   2: check<u32>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:526:2
   3: load_le<usize, u32>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:121:3
   4: load_le<usize, bitvec::order::Lsb0, u32>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:487:3
   5: init
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/architecture/riscv/dtm/jtag_dtm.rs:179:35
   6: enter_debug_mode
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/architecture/riscv/communication_interface.rs:502:9
   7: attach_jtag
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/session.rs:390:25
   8: new
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/session.rs:180:13
   9: attach_with_registry<probe_rs::config::target::TargetSelector>
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/probe.rs:349:9
  10: attach<probe_rs::config::target::TargetSelector>
             at /home/runner/work/probe-rs/probe-rs/probe-rs/src/probe.rs:334:9
  11: attach_session
             at /home/runner/work/probe-rs/probe-rs/probe-rs-tools/src/bin/probe-rs/util/common_options.rs:307:13
  12: {async_fn#0}
             at /home/runner/work/probe-rs/probe-rs/probe-rs-tools/src/bin/probe-rs/rpc/functions/probe.rs:234:23
  13: {async_fn#0}
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/postcard-rpc-0.11.9/src/server/dispatch_macro.rs:182:29
  14: {async_fn#0}<probe_rs::rpc::transport::memory::WireTx<tokio::sync::mpsc::bounded::Sender<alloc::vec::Vec<u8, alloc::alloc::Global>>>, probe_rs::rpc::transport::memory::WireRx<tokio::sync::mpsc::bounded::Receiver<core::result::Result<alloc::vec::Vec<u8, alloc::alloc::Global>, postcard_rpc::server::WireRxErrorKind>>>, alloc::boxed::Box<[u8], alloc::alloc::Global>, probe_rs::rpc::functions::impls::RpcApp<2>>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/postcard-rpc-0.11.9/src/server/mod.rs:457:33
  15: {async_block#0}
             at /home/runner/work/probe-rs/probe-rs/probe-rs-tools/src/bin/probe-rs/main.rs:510:63
[... 111 lines cut ...]
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/task/coop/mod.rs:133:5
   8: block_on<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/park.rs:284:31
   9: block_on<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/context/blocking.rs:66:9
  10: {closure#0}<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/scheduler/multi_thread/mod.rs:87:22
  11: enter_runtime<tokio::runtime::scheduler::multi_thread::{impl#0}::block_on::{closure_env#0}<probe_rs::main::{async_block_env#0}>, core::result::Result<(), anyhow::Error>>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/context/runtime.rs:65:16
  12: block_on<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/scheduler/multi_thread/mod.rs:86:9
  13: block_on_inner<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/runtime.rs:358:50
  14: block_on<probe_rs::main::{async_block_env#0}>
             at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/tokio-1.45.0/src/runtime/runtime.rs:330:13
  15: main
             at /home/runner/work/probe-rs/probe-rs/probe-rs-tools/src/bin/probe-rs/main.rs:519:5
  16: call_once<fn() -> core::result::Result<(), anyhow::Error>, ()>
             at /rustc/17067e9ac6d7ecb70e50f92c1944e545188d2359/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

### Operating System

Linux (Fedora)

albertskog, 2025-07-14:
The problem seems to be introduced with ff8cace3d4e57dc0bf68f9bd453b8ee7ddf28f71, specifically, `probe-rs/src/probe/wlink/mod.rs:439`:

```diff
- Ok(0x71_u32.to_le_bytes().to_vec())
+ Ok(0x71_usize.view_bits().to_bitvec())
```

I think `usize` will be 64 bits like on the host, but the field needs to be 32 bits like on the target? Is there a reason to change from `u32` to `usize`? I suppose the type needs to reflect the target system and not the host?

acedogblast, 2025-08-07:
I also have this issue when using probe-rs with just the `info` command. 

```
./probe-rs info
Probing target via JTAG
-----------------------

thread 'tokio-runtime-worker' panicked at /home/trc/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bitvec-1.0.1/src/field.rs:526:2:
cannot load 32 bits from a 64-bit region
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
 WARN postcard_rpc::host_client::util: in_worker: wire receive error, exiting
 WARN probe_rs::rpc::client: Failed to read topic
Error while probing target: Connection closed
No DAP interface was found on the connected probe. ARM specific information cannot be printed.

Probing target via SWD
----------------------

Error while probing target: Failed to subscribe to 'info/data': IoClosed

thread 'main' panicked at probe-rs-tools/src/bin/probe-rs/main.rs:517:22:
called `Result::unwrap()` on an `Err` value: JoinError::Panic(Id(17), "cannot load 32 bits from a 64-bit region", ...)

```

wlink works just fine.

albertskog, 2025-10-15:
Yes, looks like it was fixed already with #3366