probe-rs/probe-rs #3477: Add Example to Demonstrate Multi-Core Setup

Tomer-Eliahu, 2025-07-19:
I am working with a nRF5340 chip and am having trouble understanding how to best setup probe-rs.

As the application core and networking core have different targets and different memory.x files, 
I currently have each core setup in a separate directory (with the idea being flashing one core at a time).

However, while probe-rs works fine for the application core, it doesn't work for the network core.
Interestingly, I have no problem flashing the network core using the cargo embed command (it is just 
debugging with probe-rs that fails). 

The main issue is that flashing the network core seems to erase the application core code.

I would appreciate some help in understanding how to setup this system.
I think an example of what the intended setup should be somewhere in the docs will help others as well.
In particular, (from this release)[https://probe.rs/blog/0110-release/], it seems like you made it work but don't specify how.

Tomer-Eliahu, 2025-07-20:
I think I figured it out. I will reopen this issue if it turns out I didn't.

closed 2025-07-20 by Tomer-Eliahu

Yatekii, 2025-07-20:
It would be nice if you extend the docs and make it easier to find so others don't have the same issue :)