probe-rs/probe-rs #3480: Unable to flash rp2040 with current master and cmsis-dap

andrewleech, 2025-07-22:
### Describe the bug

Flashing a rpi-pico with a generic cmsis-dap programmer is failing for me with builds on current master.
``` bash
[corona@NUG thread]$ ~/probe-rs/target/release/probe-rs download -i /home/corona/mpy/thread/ports/rp2/build-RPI_PICO_W/firmware.elf --chip rp2040
ERROR probe_rs::vendor::raspberrypi::sequences::rp2040: Unable to connect to default address
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(1002927): An error occurred in the communication with an access port or debug port.
ERROR probe_rs::vendor::raspberrypi::sequences::rp2040: Unable to connect to default address
 WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.
 WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(1002927): An error occurred in the communication with an access port or debug port.
ERROR probe_rs::vendor::raspberrypi::sequences::rp2040: Unable to connect to default address
 WARN probe_rs::session: Failed to deconfigure device during shutdown: Arm(Dap(NoAcknowledge))
 WARN probe_rs::architecture::arm::communication_interface: Failed to stop DP Multidrop(1002927)
 WARN probe_rs::architecture::arm::communication_interface: Failed to stop DP Multidrop(11002927)
Error: An error with the flashing procedure has occurred.

Caused by:
    0: Failed to reset, and then halt the CPU.
    1: An ARM specific error occurred.
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.
```

When run with latest release (0.29.1)
``` bash
[corona@NUG thread]$ probe-rs download /home/corona/mpy/thread/ports/rp2/build-RPI_PICO_W/firmware.elf --chip rp2040
      Erasing ✔ 100% [####################] 852.00 KiB @  78.77 KiB/s (took 11s)
  Programming ✔ 100% [####################] 852.00 KiB @  24.34 KiB/s (took 35s)                                                                                                                                 
     Finished in 45.83s
```

### To Reproduce

1. Connect a cmsis-dap debugger to a rpi-pico
2. Flash elf file from suitable project (in my case micropython) with current release probe-rs
3. compile master build and repeat

### Stacktrace

```shell
Run with full debug log enabled: 
[probe-rs-debug.log](https://github.com/user-attachments/files/21356894/probe-rs-debug.log)
```

### Operating System

Linux (Ubuntu)

### Additional context

I've bisected the change in behaviour to https://github.com/probe-rs/probe-rs/pull/3429 before this it works, after the error as above

bugadani, 2025-07-22:
@xobs that reset PR didn't turn out to be very successful :/

Wren6991, 2025-07-23:
If it's failing only for some people it's possible it's timing-related. Looking at #3429:

* The attention sequence for DP 0 seems to be issued a maximum of 5 times, immediately after rescue reset removal.
    * There is some time delay between the rescue reset deasserting and the core DPs coming out of reset (because the ROSC needs to start up to provide a clock for the reset synchronisers in the PSM)
    * It's conceivable this is longer than the time to get NAK'd 5 times
    * As an experiment, maybe crank the retry count in `DefaultArmDebugSequence.debug_port_setup()`, or reduce SWCLK frequency
* Probably worth explicitly clearing the RESCUE status flag once the core Mem-APs are available again.
    * The bootrom will do this if it's left to run for long enough, but if not, the _next_ launch will see the flag and halt.
    * See ROM source: https://github.com/raspberrypi/pico-bootrom-rp2040/blob/ef22cd8ede5bc007f81d7f2416b48db90f313434/bootrom/bootrom_rt0.S#L245-L260
    * This doesn't match OP's trace but it's a potential trap

Apologies if I've missed anything as I'm not that familiar with Rust or probe-rs. I was pointed here by @jonathanpallant

xobs, 2025-07-23:
Thanks for the input, @Wren6991.

I'd like to replicate this. Do you see the same failure with the repo I posted above? If not, can you post a repo that I can build that reproduces the issue?

I'd need two things to replicate the issue:

1. A failing repo that I can build. Can you test the repo above?
2. The same (or similar) probe to what you have. Which probe are you using?

andrewleech, 2025-07-23:
Ah sorry for the confusion with the `-i` I picked the wrong cli line to copy (-i is a feature under development branched off current master). I'll double-check that the debug log was for the correct command / regenerate.

I've only been using `download`, not `run`. My firmware of interest is micropython, a binary of which can be downloaded for test:
``` bash
wget https://micropython.org/resources/firmware/RPI_PICO2_W-20250717-v1.26.0-preview.378.gc72a3e528.uf2
probe-rs download --chip rp2040 --binary-format uf2 RPI_PICO2_W-20250717-v1.26.0-preview.378.gc72a3e528.uf2
```

I'm programming with just swd and sck pins, no hardware reset connected. 
These particular tests have been with a [WCH-LinkE](https://www.olimex.com/Products/RISC-V/WCH/WCH-LinkE/) in cmsis-dap mode which has worked for me with any cmsis-dap / daplink software. I've been testing this debugger for its high-speed usb (and a [usb-c version of it](https://www.aliexpress.com/item/1005005983875152.html)). I'll try to get another rpi pico flashed as daplink to test with that too.

xobs, 2025-07-23:
That's interesting. The PICO2_W firmware works on the RP2040 Pico?

I'll see if I can't get that USB-C version of the probe, but I'm going to be unavailable for the next three weeks and I don't think it can get here in time.

It may very well be a timing issue. In which case you might want to try the fixes suggested in https://github.com/probe-rs/probe-rs/issues/3480#issuecomment-3105226251

andrewleech, 2025-07-23:
Ah whoops again, wrong download link :-/ both firmwares will download but no, the PICO2_W firmware didn't actually run on the PICO_W I had plugged in.

xobs, 2025-08-19:
I thought I'd gotten the USB-C variant, but I got the USB-A version instead. Still, it exhibits the same issue.

From my understanding, the WCH-Link is buggy. If you try to access the last multidrop address (where the Recovery AP is) then the WCH-Link locks up completely. I'm guessing that particular CMSIS-DAP device isn't capable of doing the recommended recovery sequence. I'm seeing issues with other debug software when using a WCH-Link with multidrop, even when running the latest firmware on the probe.

I did find a workaround, however. If you read the value back after writing it, the probe does not lock up.I'll submit a PR with that change.

andrewleech, 2025-08-19:
Oh interesting, thanks for the investigation and workaround!  I'd run into a couple of other issues with this probe strugglying with targets like mimxrt1176
I've since reflashed mine with https://github.com/prosper00/CH32V305-DAPLink-HS which makes it much more stable and _faster_

I didn't have any luck programming it on with the built in bootloader, not sure if I made a mistake or if the built in one got flashed over. Either way I recovered by flashing with a second wch-link - just wire gnd, 3v, , swd and scl together on both probes and plug the second one into computer to program the first.

PR #3530 (2025-08-20): rp2040: read back value of rescue dp after writing it
Read back the value of the RESCUE_DP after writing it. This works around an issue on WCH-Link probes where not doing this causes the probe to lock up.

Additionally, it was discovered that the `debug_port_setup()` implementation was breaking things on this probe. The comment said "This works around an issue where the multidrop DP becomes unavailable.", however it didn't say what the issues were. If the problem resurfaces, this code will be brought back and the comment will be updated with the actual issue it solves.

Closes #3480.