probe-rs/probe-rs #3495: No info or debug messages on STM32H755CM4 core

zinkkkk, 2025-07-28:
### Describe the bug

I am unable to get any info or debug messages when running code on STM32H755CM4 Cortex-M4 core of STM32H755ZI. The program appear to run fine other than no debug or rtt info message output. It flashes successfully but stays as Finished line with no further output.

Running code on CM7 runs as normal and outputs info & debug statements as expected.

I am using NUCLEO-H755ZI-Q dev board over usb cable to inbuilt ST-LinkV3.
Just trying the embassy examples either blinky or intercore or a heavily stripped down version of either to do nothing other than a print statement still does not fix the issue.
Code from: [embassy examples stm32h755cm7 / stm32h755cm4](https://github.com/embassy-rs/embassy/tree/main/examples/)

Using Windows 10 22H2

Running the ST-Link utility program i can access data from both cores using either Access port 0 (CM7) or 3 (CM4) 

Running probe-rs info does show a few warning, not sure if this is related
```
Probing target via JTAG
-----------------------

 INFO attach: probe_rs::probe::stlink: Target voltage (VAPP): 3.27 V
 WARN attach: probe_rs::probe::stlink: send_jtag_command 242 failed: JtagNoDeviceConnected
 INFO probe_rs::rpc::client: Received stop
Failed to identify target using protocol JTAG: An error with the usage of the probe occurred

Caused by:
    0: An error which is specific to the debug probe in use occurred.
    1: Command failed with status JtagNoDeviceConnected.

Probing target via SWD
----------------------

 INFO attach: probe_rs::probe::stlink: Target voltage (VAPP): 3.27 V
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0001000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR1 has invalid preamble (expected 0x0, got 0x1)
 WARN probe_rs::architecture::arm::memory::romtable: Component at 0xe0000000: CIDR2 has invalid preamble (expected 0x5, got 0xb1)
 INFO probe_rs::rpc::client: Received stop
ARM Chip with debug port Default:

Debug Port: DPv2, Designer: STMicroelectronics, Part: 0x4500, Revision: 0x0, Instance: 0x00
├── V1(0) MemoryAP
│   └── 0 MemoryAP (AmbaAhb3)
│       ├── 0xe00fe000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00ff000 ROM Table (Class 1), Designer: ARM Ltd
│       ├── 0xe0041000 ETM architecture (Coresight Component)
│       └── 0xe0043000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
├── V1(1) MemoryAP
│   └── 1 MemoryAP (AmbaAhb3)
│       └── Error during access: Another ARM error occurred: debug entry not present
├── V1(2) MemoryAP
│   └── 2 MemoryAP (AmbaApb2Apb3)
│       ├── 0xe00e0000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00e3000 CoreSight SWO   (Coresight Component)
│       ├── 0xe00e4000 CoreSight TraceFunnel (Coresight Component)
│       ├── 0xe00e5000 System TSGEN    (Core Link / Prime Cell / System component)
│       ├── 0xe00f0000 ROM Table (Class 1), Designer: STMicroelectronics
│       ├── 0xe00f1000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
│       ├── 0xe00f3000 CoreSight TraceFunnel (Coresight Component)
│       ├── 0xe00f4000 CoreSight TMC   (Coresight Component)
│       └── 0xe00f5000 CoreSight TPIU  (Coresight Component)
└── V1(3) MemoryAP
    └── 3 MemoryAP (AmbaAhb3)
        ├── 0xe00ff000 ROM Table (Class 1), Designer: STMicroelectronics
        ├── 0xe0001000 Generic
        ├── 0xe0000000 Peripheral test block
        ├── 0xe0041000 Cortex-M4 ETM   (Coresight Component)
        └── 0xe0043000 Coresight Component, Part: 0x0906, Devtype: 0x14, Archid: 0x0000, Designer: ARM Ltd
```

I have tried:
Running on a second computer with fresh install of everything
Updated ST-Link driver (STSW-LINK009)
Updated ST-Link firmware to latest (V3.J7.M2, STM32 Debug+Mass storage+VCP, HSI/2 8Mhz)
Removing defmt, panic probe and println!("cargo:rustc-link-arg-bins=-Tdefmt.x"); from build.rs on cm7 section

According to notes on other software packages it should be possible to get messages from either or both
[AN5286](https://www.st.com/resource/en/application_note/an5286-stm32h7x5x7-dualcore-microcontroller-debugging-stmicroelectronics.pdf)
[AN5361](https://www.st.com/resource/en/application_note/an5361-getting-started-with-projects-based-on-dualcore-stm32h7-microcontrollers-in-stm32cubeide-stmicroelectronics.pdf)

Not sure what else i can add here or if missing something obvious, obviously happy to test any ideas.

### Stacktrace

```shell

```

### Operating System

Windows

sourcebox, 2025-09-08:
I can confirm that the CM4 does not show any RTT messages in probe-rs whereas it works with OpenOCD. Adding the `--rtt-scan-memory` also does not change anything.

kwalker-plxs, 2025-09-09:
I am using an H757XI, but observed the same issue. I found that downgrading to v0.25.0 made it work for me (note v0.26.0 did not). 

I was led to this solution by looking at #3045.

jseketa, 2026-11-08:
I can reproduce this on a Nucleo-H755ZI-Q. There are two causes.

1. `run` resets the CM7, which resets the chip, so the CM4 starts and writes its RTT control block, and probe-rs clears it after that. This happens wherever the CM4 keeps its RAM. #3925 fixed the same thing for the DAP server; #4663 does it for `run` and `test`.
2. If the CM4 program keeps its RAM at 0x1000_0000, as embassy's `stm32h755cm4` examples do: the H745/H747/H755/H757 memory maps had that region as `RAM_D2` until #3036 (0.27.0) and have not had it since, so RTT does not attach. #4662 restores it.

embassy `stm32h755cm4` blinky, CM7 blinky in bank 1:

| | `attach` after a reset | `run` |
|---|---|---|
| 0.26.0 | output | no output |
| 0.32.0, master | no output | no output |
| both PRs | output | output |

(0.26.0 and 0.32.0: a copy of the example with `.ram_d3` marked NOLOAD, otherwise the same.)

With RAM at 0x3000_0000 instead, `attach` after a reset works on 0.32.0, and `run` needs #4663.

Until then, for RAM at 0x1000_0000, `attach` after a reset works with a target file that has this region in the variant's `memory_map`, passed with `--chip-description-path`:

```yaml
  - !Ram
    name: RAM_D2
    range: {start: 0x10000000, end: 0x10048000}
    cores: [cm7, cm4]
```

bugadani, 2026-11-08:
> run resets the CM7, which resets the chip, so the CM4 starts and writes its RTT control block

I think there's the bug. Halt the second core before it does anything important.

jseketa, 2026-11-08:
> Halt the second core before it does anything important.

With embassy's dual-core start-up, that works only if the CM4 is resumed first. Nucleo-H755ZI-Q, embassy `stm32h755cm7` and `stm32h755cm4` blinky. Both cores caught at a reset (VC_CORERESET), the CM4's RTT ID cleared, both resumed; the CM4's LED watched for 3 s. 5 rounds each:

| | CM4 program |
|---|---|
| plain reset, no catch | LED blinks, 5 of 5 |
| CM4 resumed, CM7 about 12 ms later | LED blinks, 5 of 5 |
| CM7 resumed, CM4 about 12 ms later | LED never changes, 5 of 5 |
| CM7 resumed, CM4 1 s later | LED never changes, 5 of 5 |

`init_primary` on the CM7 signals the CM4 through a hardware semaphore, and the CM4 clears that signal before it starts waiting for it. A CM4 that starts after the signal waits forever; in the ten rounds where the LED never changes, HSEM C1ISR is 0x1 (the CM7 signalled) and C2ISR 0x0 (cleared by the CM4). That is a race in embassy, but a debugger that holds the CM4 runs into it. `probe-rs run` on master does, with a target file that has the 0x1000_0000 region on `cm4` only: it resets and halts the CM4, and the CM4 program prints one line, 3 of 3; halted afterwards, it is in `init_secondary`.

#4663 clears before the reset instead, as the DAP server does since #3925. With #4662's region on both cores, `run` then resets and halts the CM7, so the CM4 starts first. Which would you prefer?