probe-rs/probe-rs #3564: ESP32-S3 doesn't boot after flash, then fails with 'erase_sector' failed with code 65536

0x53A, 2025-09-21:
### Describe the bug

The error messages are similar to https://github.com/probe-rs/probe-rs/issues/2338, but I'm still getting this on the latest master branch.

My project is an embassy no_std project, so no IDF.

```
> probe-rs --version
probe-rs 0.29.0 (git commit: 6665ba11)
```

```
> probe-rs info
Probing target via JTAG
-----------------------

RISC-V Chip:
  IDCODE: 0000000000
    Version:      0
    Part:         0
    Manufacturer: 0 (Unknown Manufacturer Code)

Xtensa Chip:
  IDCODE: 00120034e5
    Version:      1
    Part:         8195
    Manufacturer: 626 (Tensilica)
```

```
> espflash board-info  
[2025-09-21T03:47:58Z INFO ] Serial port: 'COM14'
[2025-09-21T03:47:58Z INFO ] Connecting...
[2025-09-21T03:47:59Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       94:a9:90:2e:93:64
```

With this in my config.toml:

```
runner = "probe-rs run --chip esp32s3 --probe=303a:1001:94:A9:90:2E:93:64 --idf-partition-table=partitions.csv"
```

The first time I flash, flash succeeds, but program doesn't start (it should blink a LED, and the exact same binary flashed with ``espflash`` works.)

```
    Finished `release` profile [optimized + debuginfo] target(s) in 0.60s
     Running `probe-rs run --chip esp32s3 --probe=303a:1001:94:A9:90:2E:93:64 --idf-partition-table=partitions.csv target\xtensa-esp32s3-none-elf\release\esp32_partylight`
      Erasing ✔ 100% [####################] 448.00 KiB @ 252.66 KiB/s (took 2s)
  Programming ✔ 100% [####################] 210.29 KiB @  55.07 KiB/s (took 4s)                                          
     Finished in 5.91s
```

Here is the output if I flash and run with espflash:

```

     Running `espflash flash --monitor --partition-table=partitions.csv target\xtensa-esp32s3-none-elf\release\esp32_partylight`
[2025-09-21T03:52:13Z INFO ] Serial port: 'COM14'
[2025-09-21T03:52:13Z INFO ] Connecting...
[2025-09-21T03:52:14Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       94:a9:90:2e:93:64
Partition table:   partitions.csv
App/part. size:    338,480/1,048,576 bytes, 32.28%
[00:00:01] [========================================]      14/14      0x0                                                
[2025-09-21T03:52:17Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:19] [========================================]     198/198     0x10000                                            
[2025-09-21T03:52:38Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c000020 size=0d384h ( 54148) map
I (118) esp_image: segment 1: paddr=0001d3ac vaddr=3fc8f4e8 size=02c6ch ( 11372) load
I (122) esp_image: segment 2: paddr=00020020 vaddr=42010020 size=3743ch (226364) map
I (180) esp_image: segment 3: paddr=00057464 vaddr=3fc92154 size=040bch ( 16572) load
I (185) esp_image: segment 4: paddr=0005b528 vaddr=40378000 size=074e4h ( 29924) load
I (199) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
Hello, world!
[...]
```

(Hello World is printed by my firmware, anything before that is the bootloader on the chip)

If I then unplug and reconnect the device, I can't flash it with probe-rs at all:

```
> cargo run --release
[...]
    Finished `release` profile [optimized + debuginfo] target(s) in 0.52s
     Running `probe-rs run --chip esp32s3 --probe=303a:1001:94:A9:90:2E:93:64 --idf-partition-table=partitions.csv target\xtensa-esp32s3-none-elf\release\esp32_partylight`
      Erasing ✔  14% [###-----------------]  64.00 KiB @ 273.12 KiB/s (ETA 0s)                                           
Error: An error with the flashing procedure has occurred.                                                                

Caused by:
    0: Failed to erase flash sector at address 0x00010000.
    1: The execution of 'erase_sector' failed with code 65536. This might indicate a problem with the flash algorithm.   
error: process didn't exit successfully: `probe-rs run --chip esp32s3 --probe=303a:1001:94:A9:90:2E:93:64 --idf-partition-table=partitions.csv target\xtensa-esp32s3-none-elf\release\esp32_partylight` (exit code: 1)
```

I can reflash it with ``espflash``, and after flashing with espflash once, ``probe-rs`` would also flash (but not run).

---

So i believe the flash algorithm error can be ignored, as it's a follow up error. Either I or probe-rs seem to be doing something incorrect with the flash, getting the chip into an incorrect state from where it can't recover until flashed over serial again.

### Stacktrace

```shell

```

### Operating System

None

0x53A, 2025-09-21:
AHA, just after posting the issue, I thought of one other thing:

I flashed it with probe-rs, disconnected, connected the other usb port and ran ``probe-rs monitor``.

It spams this text and repeatedly resets itself, which is probably why it can't reconnect to USB JTAG.

```
[...]
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cceb5
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd24
load:0x403cb700,len:0x2f04
entry 0x403c8928
I (32) boot: ESP-IDF v5.4.1-426-g3ad36321ea 2nd stage bootloader
I (32) boot: compile time Apr 24 2025 15:55:11
I (32) boot: Multicore bootloader
I (33) boot: chip revision: v0.2
I (36) boot: efuse block revision: v1.3
I (40) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (47) boot.esp32s3: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c000020 size=0d384h ( 54148) map
E (95) boot_comm: Image requires efuse blk rev >= v179.75, but chip is v1.3
E (102) boot: Factory app partition is not bootable
E (106) boot: No bootable app partitions in the partition table
[...]
```

0x53A, 2025-09-21:
espflash and probe-rs seem to flash different second stage bootloaders, or probably it's because I am using and old version of espflash and the latest version of probe-rs.

I added

```
esp_bootloader_esp_idf::esp_app_desc!();
```

to my main.rs, now it works.

bugadani, 2025-09-21:
```
Caused by:
    0: Failed to erase flash sector at address 0x00010000.
```

I've seen this way too often, and I don't know the root cause of this issue. But it's way too annoying.

As for the app descriptor check, you're right. Probe-rs should also check that you flash a binary made for the correct device, I'll try to summon up some energy and do this in one batch.

0x53A, 2025-10-05:
> Caused by:
    0: Failed to erase flash sector at address 0x00010000.

> I've seen this way too often, and I don't know the root cause of this issue. But it's way too annoying.

This message is printed if the chip resets itself while probe-rs tries to erase/flash it.

It might be worth it to make a seperate issue for this, but currently it seems impossible to recover a chip that's in a reset loop.

In my initial case, the missing app descriptor caused a reset loop, but alternatively just flash a program with something like this

```rs
#[esp_hal_embassy::main]
async fn main(spawner: Spawner) -> ! {

    esp_hal::rom::software_reset();
}
```

At least for me, esp32-s3 under windows 11, ``probe-rs erase`` fails.

```
> probe-rs erase --connect-under-reset
      Erasing ✔ 0s                                                                                                                             
Error: The execution of 'uninit' failed with code 1. This might indicate a problem with the flash algorithm.
```

closed 2025-11-28 by MabezDev