probe-rs/probe-rs #3566: Can't detect debug probe after linux update

ValouBambou, 2025-09-22:
### Describe the bug

I noticed that after an update made on Arch Linux maybe in the previous week, probe-rs can't detect my probe (debug probe for the rp2040) anymore. I tried to update probe-rs to the latest version, but it didn't solve the issue. Maybe there is a kernel change that broke it, but I have no clue. I got the logs with trace level in case it can help to diagnostic more accurately the issue.

[probe-rs.log](https://github.com/user-attachments/files/22465559/probe-rs.log)

### Stacktrace

```shell

```

### Operating System

Other Linux Distribution

kevinmehall, 2025-09-22:
It's finding the device and attempting to open it, but then the log just goes on to the next candidate device.

```
[35mTRACE[0m [2mprobe_rs::probe::cmsisdap::tools[0m[2m:[0m Trying device DeviceInfo { bus_number: 1, device_address: 6, vendor_id: 0x2E8A, product_id: 0x000C, device_version: 0x0101, class: 0x00, subclass: 0x00, protocol: 0x00, speed: Some(Full), manufacturer_string: Some("Raspberry Pi"), product_string: Some("Debug Probe (CMSIS-DAP)"), serial_number: Some("E6614103E726A624"), sysfs_path: SysfsPath("/sys/bus/usb/devices/1-9"), interfaces: [InterfaceInfo { interface_number: 0, class: 0xFF, subclass: 0x00, protocol: 0x00, interface_string: Some("CMSIS-DAP v2 Interface") }, InterfaceInfo { interface_number: 1, class: 0x02, subclass: 0x02, protocol: 0x00, interface_string: Some("CDC-ACM UART Interface") }, InterfaceInfo { interface_number: 2, class: 0x0A, subclass: 0x00, protocol: 0x00, interface_string: None }] }
[35mTRACE[0m [2mprobe_rs::probe::cmsisdap::tools[0m[2m:[0m   Interface 0: CMSIS-DAP v2 Interface
[35mTRACE[0m [2mprobe_rs::probe::cmsisdap::tools[0m[2m:[0m interface 2 has no string, skipping
[35mTRACE[0m [2mprobe_rs::probe::cmsisdap::tools[0m[2m:[0m Debug Probe (CMSIS-DAP): CMSIS-DAP device with 3 interfaces
[35mTRACE[0m [2mprobe_rs::probe::cmsisdap::tools[0m[2m:[0m No HID interface for CMSIS-DAP found.
[34mDEBUG[0m [2mnusb::platform::linux_usbfs::enumeration[0m[2m:[0m sysfs read "/sys/bus/usb/devices/1-9/bConfigurationValue": Ok("1\n")
[34mDEBUG[0m [2mnusb::platform::linux_usbfs::device[0m[2m:[0m Opening usbfs device /dev/bus/usb/001/006
[34mDEBUG[0m [2mnusb::platform::linux_usbfs::enumeration[0m[2m:[0m probe device SysfsPath("/sys/bus/usb/devices/usb3")
```

probe-rs cmsis-dap silently ignores any errors when opening a device, so I can't tell from the log what the error is.
https://github.com/probe-rs/probe-rs/blob/6665ba11ba097cb0f06318539c61ca87e6886092/probe-rs/src/probe/cmsisdap/tools.rs#L160-L162

My guess, though, would be a permission issue, if the update removed or broke a udev rule that gives you access to the device. See https://probe.rs/docs/getting-started/probe-setup/#linux-udev-rules for probe-rs's recommended rules.

You can verify that the udev rules are working with `getfacl /dev/bus/usb/*/* `. It's `/dev/bus/usb/001/006` in your log, but that will change if the probe is re-connected.

ValouBambou, 2025-09-28:
I agree that documents this in the official documention is probably the best we can do but I'm not a systemd expert so maybe modify the udev rules file is also a possible solution.

ValouBambou, 2025-10-03:
I can test this in the next week. Just tell me which PR to test because now there are 2 of them ;)

bitbloxhub, 2026-09-06:
It seems like the `uaccess` rule may not be working on `systemd` 258+, even with the group working.
I think this may be due to using `ACTION!="add|change", GOTO="probe_rs_rules_end"`.
https://discourse.nixos.org/t/probe-rs-sees-probes-but-they-are-inaccessible/79834/11